grblHAL/LPC176x #1: compilation issues, possible makefile export branch?

naturesyouth, 2021-03-25:
hi, very excited to find this project, especially as it looks like i will technically be able to build it for my skr 1.3, however i have not been able to get it to work, the arduino library method just says no libary found, and trying to load it into a fresh version of eclipse cdt embedded wasn't able to import the project properly at all. would it be possible to do an export of a makefile setup, which should be trivial from eclipse from what i have read.

terjeio, 2021-03-25:
>it looks like i will technically be able to build it for my skr 1.3

You will, I have tested with such a board.

> the arduino library method just says no libary found

It is not made with or for the Arduino IDE/libs - and I have no idea how much work it will be to get it to work. I'll leave such an excercise to somebody else as I am no fan of the Arduino IDE.

> would it be possible to do an export of a makefile setup, which should be trivial from eclipse from what i have read.

Read where?

Using [MCUXpresso](https://www.nxp.com/design/software/development-software/mcuxpresso-software-and-tools/mcuxpresso-integrated-development-environment-ide:MCUXpresso-IDE) is not an option?

terjeio, 2021-03-25:
I have uploaded a binary for a plain 3 axis machine for you to try [here](http://www.io-engineering.com/downloads) - bootloader version.

MeSmash, 2021-04-20:
same here: I get following error in MCUxpresso - all compiled with git as per instructions. latest version of MCUXpresso.

see below

Building file: ../src/main.c
../src/driver.c: In function 'driver_setup':
../src/driver.c:1033:54: error: 'X_DISABLE_PN' undeclared (first use in this function); did you mean 'X_DISABLE_PIN'?
 1033 |     Chip_IOCON_PinMux((LPC_IOCON_T *)LPC_IOCON_BASE, X_DISABLE_PN, X_DISABLE_PIN, IOCON_MODE_INACT, IOCON_FUNC0);
      |                                                      ^~~~~~~~~~~~
      |                                                      X_DISABLE_PIN
../src/driver.c:1033:54: note: each undeclared identifier is reported only once for each function it appears in
../src/driver.c:1034:54: error: 'Y_DISABLE_PN' undeclared (first use in this function); did you mean 'Y_DISABLE_PIN'?
 1034 |     Chip_IOCON_PinMux((LPC_IOCON_T *)LPC_IOCON_BASE, Y_DISABLE_PN, Y_DISABLE_PIN, IOCON_MODE_INACT, IOCON_FUNC0);
      |                                                      ^~~~~~~~~~~~
      |                                                      Y_DISABLE_PIN
../src/driver.c:1035:54: error: 'Z_DISABLE_PN' undeclared (first use in this function); did you mean 'Z_DISABLE_PIN'?
 1035 |     Chip_IOCON_PinMux((LPC_IOCON_T *)LPC_IOCON_BASE, Z_DISABLE_PN, Z_DISABLE_PIN, IOCON_MODE_INACT, IOCON_FUNC0);
      |                                                      ^~~~~~~~~~~~
      |                                                      Z_DISABLE_PIN
Building file: ../src/mks_sbase.c
make: *** [src/subdir.mk:58: src/driver.o] Error 1
make: *** Waiting for unfinished jobs....
Invoking: MCU C Compiler
arm-none-eabi-gcc -DNDEBUG -D__CODE_RED -DCORE_M3 -D__USE_CMSIS=CMSIS_CORE_LPC17xx -D__LPC17XX__ -D__NEWLIB__ -D_USE_LPCOPEN -DNO_BOARD_LIB -I"C:\Users\mjwoo\Documents\MCUXpressoIDE_11.3.1_5262\workspace\GRBL Driver LPC176x\src" -I"C:\Users\mjwoo\Documents\MCUXpressoIDE_11.3.1_5262\workspace\GRBL Driver LPC176x\src\FatFs" -I"C:\Users\mjwoo\Documents\MCUXpressoIDE_11.3.1_5262\workspace\GRBL Driver LPC176x\src\lpc_chip_175x_6x\inc" -I"C:\Users\mjwoo\Documents\MCUXpressoIDE_11.3.1_5262\workspace\GRBL Driver LPC176x\src\lpc_chip_175x_6x\inc\usbd" -Os -fno-common -g -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmacro-prefix-map="../src/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/main.d" -MT"src/main.o" -MT"src/main.d" -o "src/main.o" "../src/main.c"
Invoking: MCU C Compiler
arm-none-eabi-gcc -DNDEBUG -D__CODE_RED -DCORE_M3 -D__USE_CMSIS=CMSIS_CORE_LPC17xx -D__LPC17XX__ -D__NEWLIB__ -D_USE_LPCOPEN -DNO_BOARD_LIB -I"C:\Users\mjwoo\Documents\MCUXpressoIDE_11.3.1_5262\workspace\GRBL Driver LPC176x\src" -I"C:\Users\mjwoo\Documents\MCUXpressoIDE_11.3.1_5262\workspace\GRBL Driver LPC176x\src\FatFs" -I"C:\Users\mjwoo\Documents\MCUXpressoIDE_11.3.1_5262\workspace\GRBL Driver LPC176x\src\lpc_chip_175x_6x\inc" -I"C:\Users\mjwoo\Documents\MCUXpressoIDE_11.3.1_5262\workspace\GRBL Driver LPC176x\src\lpc_chip_175x_6x\inc\usbd" -Os -fno-common -g -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmacro-prefix-map="../src/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/mks_sbase.d" -MT"src/mks_sbase.o" -MT"src/mks_sbase.d" -o "src/mks_sbase.o" "../src/mks_sbase.c"
Finished building: ../src/flash.c
Finished building: ../src/i2c.c
 
Finished building: ../src/main.c
 
 
Finished building: ../src/mks_sbase.c
 
"make -r -j24 all" terminated with exit code 2. Build might be incomplete.

12:01:19 Build Failed. 5 errors, 24 warnings. (took 5s.626ms)

MeSmash, 2021-04-20:
changed my.machine from the smoothieboard to the SKR 1.3 and this is the error now:

Building target: Firmware.axf
Invoking: MCU Linker
arm-none-eabi-gcc -nostdlib -L"C:\Users\mjwoo\Documents\MCUXpressoIDE_11.3.1_5262\workspace\GRBL Driver LPC176x\src\lpc_chip_175x_6x\libs" -Xlinker -Map="Firmware.map" -Xlinker --cref -Xlinker --gc-sections -Xlinker -print-memory-usage -mcpu=cortex-m3 -mthumb -T "GRBL_Driver_LPC176x_BL_0x4000.ld" -o "Firmware.axf"  ./src/trinamic/common.o ./src/trinamic/tmc2130.o ./src/trinamic/tmc2130hal.o ./src/trinamic/tmc2209.o ./src/trinamic/tmc2209hal.o ./src/trinamic/tmc26x.o ./src/trinamic/tmc5160.o ./src/trinamic/tmc5160hal.o ./src/trinamic/tmc_interface.o  ./src/sdcard/sdcard.o  ./src/motors/trinamic.o  ./src/lpc_chip_175x_6x/src/adc_17xx_40xx.o ./src/lpc_chip_175x_6x/src/can_17xx_40xx.o ./src/lpc_chip_175x_6x/src/chip_17xx_40xx.o ./src/lpc_chip_175x_6x/src/clock_17xx_40xx.o ./src/lpc_chip_175x_6x/src/cmp_17xx_40xx.o ./src/lpc_chip_175x_6x/src/crc_17xx_40xx.o ./src/lpc_chip_175x_6x/src/dac_17xx_40xx.o ./src/lpc_chip_175x_6x/src/eeprom_17xx_40xx.o ./src/lpc_chip_175x_6x/src/emc_17xx_40xx.o ./src/lpc_chip_175x_6x/src/enet_17xx_40xx.o ./src/lpc_chip_175x_6x/src/gpdma_17xx_40xx.o ./src/lpc_chip_175x_6x/src/gpio_17xx_40xx.o ./src/lpc_chip_175x_6x/src/gpioint_17xx_40xx.o ./src/lpc_chip_175x_6x/src/i2c_17xx_40xx.o ./src/lpc_chip_175x_6x/src/i2s_17xx_40xx.o ./src/lpc_chip_175x_6x/src/iap.o ./src/lpc_chip_175x_6x/src/iocon_17xx_40xx.o ./src/lpc_chip_175x_6x/src/lcd_17xx_40xx.o ./src/lpc_chip_175x_6x/src/pmu_17xx_40xx.o ./src/lpc_chip_175x_6x/src/ring_buffer.o ./src/lpc_chip_175x_6x/src/ritimer_17xx_40xx.o ./src/lpc_chip_175x_6x/src/rtc_17xx_40xx.o ./src/lpc_chip_175x_6x/src/sdc_17xx_40xx.o ./src/lpc_chip_175x_6x/src/sdmmc_17xx_40xx.o ./src/lpc_chip_175x_6x/src/spi_17xx_40xx.o ./src/lpc_chip_175x_6x/src/ssp_17xx_40xx.o ./src/lpc_chip_175x_6x/src/stopwatch_17xx_40xx.o ./src/lpc_chip_175x_6x/src/sysctl_17xx_40xx.o ./src/lpc_chip_175x_6x/src/sysinit_17xx_40xx.o ./src/lpc_chip_175x_6x/src/timer_17xx_40xx.o ./src/lpc_chip_175x_6x/src/uart_17xx_40xx.o ./src/lpc_chip_175x_6x/src/wwdt_17xx_40xx.o  ./src/lpc17xx/_exit.o  ./src/grbl-lpc/pwm_driver.o  ./src/grbl/coolant_control.o ./src/grbl/corexy.o ./src/grbl/gcode.o ./src/grbl/grbllib.o ./src/grbl/limits.o ./src/grbl/maslow.o ./src/grbl/motion_control.o ./src/grbl/my_plugin.o ./src/grbl/nuts_bolts.o ./src/grbl/nvs_buffer.o ./src/grbl/override.o ./src/grbl/pid.o ./src/grbl/planner.o ./src/grbl/protocol.o ./src/grbl/report.o ./src/grbl/settings.o ./src/grbl/sleep.o ./src/grbl/spindle_control.o ./src/grbl/state_machine.o ./src/grbl/stepper.o ./src/grbl/stream.o ./src/grbl/system.o ./src/grbl/tool_change.o ./src/grbl/wall_plotter.o  ./src/eeprom/eeprom_24AAxxx.o ./src/eeprom/eeprom_24LC16B.o  ./src/btt_skr_1.x.o ./src/cdc_desc.o ./src/cdc_vcom.o ./src/cr_startup_lpc175x_6x.o ./src/crp.o ./src/diskio.o ./src/driver.o ./src/flash.o ./src/i2c.o ./src/main.o ./src/mks_sbase.o ./src/serial.o ./src/sysinit.o ./src/usb_serial.o   -lusbd_175x_6x_lib
Memory region         Used Size  Region Size  %age Used
       MFlash512:       98224 B       496 KB     19.34%
        RamLoc32:       71816 B        32 KB    219.17%
c:/nxp/mcuxpressoide_11.3.1_5262/ide/plugins/com.nxp.mcuxpresso.tools.win32_11.3.0.202008311133/tools/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: Firmware.axf section `.bss' will not fit in region `RamLoc32'
        RamAHB32:          0 GB        32 KB      0.00%
c:/nxp/mcuxpressoide_11.3.1_5262/ide/plugins/com.nxp.mcuxpresso.tools.win32_11.3.0.202008311133/tools/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: region `RamLoc32' overflowed by 39048 bytes
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:40: Firmware.axf] Error 1
"make -r -j24 all" terminated with exit code 2. Build might be incomplete.

12:16:27 Build Failed. 3 errors, 24 warnings. (took 6s.205ms)

MeSmash, 2021-04-20:
whatever is happening it appears its either the programs used to compile or there are numerous errors in the code.

Is there no other way to compile these builds ?

terjeio, 2021-04-20:
> whatever is happening it appears its either the programs used to compile or there are numerous errors in the code.

Oops, a _grbl/config.h_ file with non-default settings has been wrongly committet - it allocated a way to large planner buffer for this MCU. There was also an incompatible (with the latest settings API) board file for MKS SBASE. Fix committed for both.

Note that I do not own a MKS SBASE board so I am not able to verify the driver.

> Is there no other way to compile these builds ?

No. It may be possible to add other ways but I have not explored that.

MeSmash, 2021-04-20:
many thanks.

I have an Sbase card however,  its the SKR boards I am maily interested in. 

(I think the majority of ppl who have the SKR boards are really interested in the GrblHal project)  

Having the low cost SKR running a laser or cnc on GrblHal is very appealing.

I think many ppl, including myself, have become tired of smoothieware

carl1961, 2021-05-17:
@terjeio  thanks for the creation of the bin files but SKR 1.3 is 1768 MCU

is Your SKR 1.3 bin file is actually for LCP1769 ?

terjeio, 2021-05-17:
There is no difference in code for LCP1768 and LCP1769. The CPU clock frequency might be different though, I should revisit that?

carl1961, 2021-05-17:
@terjeio , You might need too.  I got your LPC176x to compile and i will test too. I have both SKR 1.3 and 1.4 and have a laser engraver I built to test with.

I used Git Hub desktop to clone your LPC176x then in MCUXpresso it lets you import the Git repository , then I had to fix the paths to the files lpc_chip_175x_6x by editing the paths, that removed the errors and compiled.

carl1961, 2021-05-18:
@terjeio Your SKR v1.3 plain LPC1769 driver for bootloader.zip worked, and I could get commands in terminal console.  the one I compiled did not work, computer did not recognize my board. at your convince can you share how you got your bin file setup to get the computer to see the skr 1.3.? thanks,

terjeio, 2021-05-18:
Did you compile the BL_0x4000 version as described in the [readme](https://github.com/grblHAL/LPC176x#readme) and use the corresponding binary?

carl1961, 2021-05-18:
I did not do that, Thanks the firmware.bin was in the debug folder. I may still have not setup things right. Thanks for you fast reply

terjeio, 2021-05-18:
> Thanks the firmware.bin was in the debug folder.

Are you sure? I do not get a firmware.bin file in the debug folder when I compile the debug version. When I compile the bootloader version firmware.bin in the BL_0x4000 folder is created (or updated).

carl1961, 2021-05-18:
that worked, I stared all over again, but this time i used url to clone and did as you said selected build BL_0x4000 and works fine. , I will setup my laser engraver machine and let you know how things go, Thanks so much for you  work!

closed 2021-07-26 by terjeio