grblHAL/LPC176x #2: No serial access

hcremer, 2021-07-08:
I'm not a developer and only used the Arduino IDE until now to compile and upload GRBL.
Compiled GRBLHAL with MCUXpresso IDE v11.3.1 [Build 5262] [2021-04-02] under Ubuntu 20.04. After changing the MCU to LPC1769 and the path in main.c to grbl/grbllib.h it compiled without error but a few warnings. Flashing the BTT SKR 1.4 TURBO worked but I'm not getting USB-Serial Connection to the board.
When flashing the binary provided in Issue #1, I'm getting USB-Serial connection.
What am I doing wrong?

terjeio, 2021-07-08:
> After changing the MCU to LPC1769 and the path in main.c to grbl/grbllib.h it compiled without error but a few warnings.

Oops, the path in main.c has been wrong for a long time, hidden for me by the Windows file system - I'll fix that.
The warnings are from flaws in the NXP provided drivers and can be ignored.

How do you program? Via SD card and a bootloader image or via a programmer connected to the ICSP header? I you compile and program the wrong binary it will not work.

I do not have a SKR v1.4 to test with, however the latest code works with a SKR v1.3 so it should work with the 1.4 as well.

hcremer, 2021-07-08:
Thanks for the quick reply. I selected the BL_0x4000 from the hammer icon to compile the firmware.bin file and copied it to the SD card to transfer it to the board.

terjeio, 2021-07-08:
Ok, then maybe it is due to your change of MCU messing up the USB clock?
I have uploaded a plain driver that I just compiled [here](http://www.io-engineering.com/downloads), please try that and report back.

hcremer, 2021-07-08:
The uploaded bin file worked.
I then deleted the entire workspace and reconfigured and compiled everything again without changing the MCU. The resulting bin file also worked (only tested USB serial communication, not interaction with machine and bCNC).
The two files have the same size, but not the same MD5SUM.
As a non-expert, I have two more questions: Does it matter if you leave the setting in the IDE on LPC1768 for a board with LPC1769 CPU? What does "linked against LPCOpen development platform libraries" mean (I'm using the naked MCUXpresso IDE and thought that MCUXpresso had replaced LPCOpen)?

terjeio, 2021-07-09:
> Does it matter if you leave the setting in the IDE on LPC1768 for a board with LPC1769 CPU?

No, but I do not know what was changed when you switched to LPC1769. AFAIK LPC1769 supports a higher CPU clock frequency - and changing MCU affects how the clock tree is set up? The USB peripheral has to be clocked at 48 MHz (within tight tolerances), if outside it won't enumerate. My best guess is that some constants were changed that lead to it be outside tolerance limits. The library used is for the 17xx family of processors so there is no difference in executable code.

> What does "linked against LPCOpen development platform libraries" mean (I'm using the naked MCUXpresso IDE and thought that MCUXpresso had replaced LPCOpen)?

The LPC176x family is fairly old now, the [library code](https://www.nxp.com/design/microcontrollers-developer-resources/lpcopen-libraries-and-examples/lpcopen-software-development-platform-lpc17xx:LPCOPEN-SOFTWARE-FOR-LPC17XX) used for this driver is from 2014 when it was called LPCOpen. I have not checked if there is a newer MCUXpresso SDK available and as long as the current code works I will not spend any time on that.

hcremer, 2021-07-09:
Thank you very much for your help and the great job you are doing by developing this software :-)

closed 2021-07-26 by terjeio