grblHAL/LPC176x #4: driver LPC176x i can't compile with mcuxpresso ide

silvanopas, 2021-07-26:
Hello
I can't fill in and it gives me the following messages, am I doing something wrong?

11:19:24 **** Build of configuration Debug for project LPC176x ****
make -r -j4 all 
Building file: ../src/trinamic/tmc2209.c
Building file: ../src/trinamic/tmc2130hal.c
Building file: ../src/trinamic/tmc2130.c
Building file: ../src/trinamic/common.c
Invoking: MCU C Compiler
Invoking: MCU C Compiler
Invoking: MCU C Compiler
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc2130hal.d" -MT"src/trinamic/tmc2130hal.o" -MT"src/trinamic/tmc2130hal.d" -o "src/trinamic/tmc2130hal.o" "../src/trinamic/tmc2130hal.c"
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc2130.d" -MT"src/trinamic/tmc2130.o" -MT"src/trinamic/tmc2130.d" -o "src/trinamic/tmc2130.o" "../src/trinamic/tmc2130.c"
Invoking: MCU C Compiler
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc2209.d" -MT"src/trinamic/tmc2209.o" -MT"src/trinamic/tmc2209.d" -o "src/trinamic/tmc2209.o" "../src/trinamic/tmc2209.c"
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/common.d" -MT"src/trinamic/common.o" -MT"src/trinamic/common.d" -o "src/trinamic/common.o" "../src/trinamic/common.c"
Finished building: ../src/trinamic/tmc2130.c
 
Finished building: ../src/trinamic/tmc2130hal.c
 
Building file: ../src/trinamic/tmc2209hal.c
Invoking: MCU C Compiler
Building file: ../src/trinamic/tmc26x.c
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc2209hal.d" -MT"src/trinamic/tmc2209hal.o" -MT"src/trinamic/tmc2209hal.d" -o "src/trinamic/tmc2209hal.o" "../src/trinamic/tmc2209hal.c"
Finished building: ../src/trinamic/common.c
Invoking: MCU C Compiler
 
Finished building: ../src/trinamic/tmc2209.c
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc26x.d" -MT"src/trinamic/tmc26x.o" -MT"src/trinamic/tmc26x.d" -o "src/trinamic/tmc26x.o" "../src/trinamic/tmc26x.c"
 
Building file: ../src/trinamic/tmc5160.c
Invoking: MCU C Compiler
Building file: ../src/trinamic/tmc5160hal.c
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc5160.d" -MT"src/trinamic/tmc5160.o" -MT"src/trinamic/tmc5160.d" -o "src/trinamic/tmc5160.o" "../src/trinamic/tmc5160.c"
Invoking: MCU C Compiler
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc5160hal.d" -MT"src/trinamic/tmc5160hal.o" -MT"src/trinamic/tmc5160hal.d" -o "src/trinamic/tmc5160hal.o" "../src/trinamic/tmc5160hal.c"
Finished building: ../src/trinamic/tmc2209hal.c
Finished building: ../src/trinamic/tmc26x.c
 
 
Building file: ../src/trinamic/tmc_interface.c
Building file: ../src/sdcard/sdcard.c
Invoking: MCU C Compiler
Invoking: MCU C Compiler
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../src/sdcard/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/sdcard/sdcard.d" -MT"src/sdcard/sdcard.o" -MT"src/sdcard/sdcard.d" -o "src/sdcard/sdcard.o" "../src/sdcard/sdcard.c"
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc_interface.d" -MT"src/trinamic/tmc_interface.o" -MT"src/trinamic/tmc_interface.d" -o "src/trinamic/tmc_interface.o" "../src/trinamic/tmc_interface.c"
Finished building: ../src/trinamic/tmc5160.c
 
Finished building: ../src/trinamic/tmc5160hal.c
Building file: ../src/sdcard/ymodem.c
In file included from ../src/sdcard/sdcard.c:22:
../src/sdcard/sdcard.h:30:10: fatal error: driver.h: No such file or directory
   30 | #include "driver.h"
      |          ^~~~~~~~~~
compilation terminated.
 
Invoking: MCU C Compiler
Building file: ../src/motors/trinamic.c
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../src/sdcard/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/sdcard/ymodem.d" -MT"src/sdcard/ymodem.o" -MT"src/sdcard/ymodem.d" -o "src/sdcard/ymodem.o" "../src/sdcard/ymodem.c"
Invoking: MCU C Compiler
make: *** [src/sdcard/subdir.mk:23: src/sdcard/sdcard.o] Error 1
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../src/motors/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/motors/trinamic.d" -MT"src/motors/trinamic.o" -MT"src/motors/trinamic.d" -o "src/motors/trinamic.o" "../src/motors/trinamic.c"
make: *** Waiting for unfinished jobs....
In file included from ../src/sdcard/ymodem.c:28:
../src/sdcard/sdcard.h:30:10: fatal error: driver.h: No such file or directory
   30 | #include "driver.h"
      |          ^~~~~~~~~~
compilation terminated.
../src/motors/trinamic.c:25:10: fatal error: driver.h: No such file or directory
   25 | #include "driver.h"
      |          ^~~~~~~~~~
compilation terminated.
make: *** [src/sdcard/subdir.mk:23: src/sdcard/ymodem.o] Error 1
Finished building: ../src/trinamic/tmc_interface.c
make: *** [src/motors/subdir.mk:20: src/motors/trinamic.o] Error 1
 
"make -r -j4 all" terminated with exit code 2. Build might be incomplete.

11:19:28 Build Failed. 7 errors, 0 warnings. (took 3s.391ms)

terjeio, 2021-07-26:
I got a bunch of errors when compiling for the bootloader or release versions but not the same as yours. I have updated the repo to fix the errors I got. Can you download/update the source again to see if the errors are still there? If they are then report back.

silvanopas, 2021-07-26:
Hello
If I can in the day I'll try and I'll let you know, thanks

silvanopas, 2021-07-26:
I state that I downloaded and without touching anything I filled in, the
messages are these:

16:22:40 **** Build of configuration Debug for project LPC176x ****
make -r -j4 all
Building file: ../src/trinamic/common.c
Building file: ../src/trinamic/tmc2130.c
Building file: ../src/trinamic/tmc2130hal.c
Building file: ../src/trinamic/tmc2209.c
Invoking: MCU C Compiler
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__
-D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c
-fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections
-fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3
-mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc2130.d"
-MT"src/trinamic/tmc2130.o" -MT"src/trinamic/tmc2130.d" -o
"src/trinamic/tmc2130.o" "../src/trinamic/tmc2130.c"
Invoking: MCU C Compiler
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__
-D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c
-fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections
-fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3
-mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/common.d"
-MT"src/trinamic/common.o" -MT"src/trinamic/common.d" -o
"src/trinamic/common.o" "../src/trinamic/common.c"
Invoking: MCU C Compiler
Invoking: MCU C Compiler
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__
-D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c
-fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections
-fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3
-mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc2130hal.d"
-MT"src/trinamic/tmc2130hal.o" -MT"src/trinamic/tmc2130hal.d" -o
"src/trinamic/tmc2130hal.o" "../src/trinamic/tmc2130hal.c"
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__
-D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c
-fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections
-fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3
-mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc2209.d"
-MT"src/trinamic/tmc2209.o" -MT"src/trinamic/tmc2209.d" -o
"src/trinamic/tmc2209.o" "../src/trinamic/tmc2209.c"
Finished building: ../src/trinamic/common.c

Finished building: ../src/trinamic/tmc2209.c
Building file: ../src/trinamic/tmc2209hal.c
Invoking: MCU C Compiler

arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__
-D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c
-fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections
-fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3
-mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc2209hal.d"
-MT"src/trinamic/tmc2209hal.o" -MT"src/trinamic/tmc2209hal.d" -o
"src/trinamic/tmc2209hal.o" "../src/trinamic/tmc2209hal.c"
Finished building: ../src/trinamic/tmc2130.c
Finished building: ../src/trinamic/tmc2130hal.c
Building file: ../src/trinamic/tmc26x.c

Invoking: MCU C Compiler
Building file: ../src/trinamic/tmc5160.c
Building file: ../src/trinamic/tmc5160hal.c
arm-none-eabi-gcc -DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__
-D__USE_LPCOPEN -DNO_BOARD_LIB -O0 -fno-common -g3 -Wall -c
-fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections
-fmerge-constants -fmacro-prefix-map="../src/trinamic/"=. -mcpu=cortex-m3
-mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/trinamic/tmc26x.d"
-MT"src/trinamic/tmc26x.o" -MT"src/trinamic/tmc26x.d" -o
"src/trinamic/tmc26x.o" "../src/trinamic/tmc26x.c"
Invoking: MCU C Compiler
Invoking: MCU C Compiler
[... 60 lines cut ...]
-fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections
-fmerge-constants -fmacro-prefix-map="../src/motors/"=. -mcpu=cortex-m3
-mthumb -D__NEWLIB__ -fstack-usage -MMD -MP -MF"src/motors/trinamic.d"
-MT"src/motors/trinamic.o" -MT"src/motors/trinamic.d" -o
"src/motors/trinamic.o" "../src/motors/trinamic.c"
make: *** [src/sdcard/subdir.mk:23: src/sdcard/sdcard.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from ../src/sdcard/ymodem.c:28:
../src/sdcard/sdcard.h:30:10: fatal error: driver.h: No such file or
directory
   30 | #include "driver.h"
      |          ^~~~~~~~~~
compilation terminated.
make: *** [src/sdcard/subdir.mk:23: src/sdcard/ymodem.o] Error 1
../src/motors/trinamic.c:25:10: fatal error: driver.h: No such file or
directory
   25 | #include "driver.h"
      |          ^~~~~~~~~~
compilation terminated.
Finished building: ../src/trinamic/tmc_interface.c
make: *** [src/motors/subdir.mk:20: src/motors/trinamic.o] Error 1

"make -r -j4 all" terminated with exit code 2. Build might be incomplete.

16:22:43 Build Failed. 7 errors, 0 warnings. (took 2s.420ms)

terjeio, 2021-07-26:
Is the project imported correctly? Here is how it shows up in the Project Explorer for me:

[image]

silvanopas, 2021-07-26:
To download it I used this instruction:

git clone --recurse-submodules https://github.com/grblHAL/LPC176x.git

To import it into MCUCpresso IDE I used:

import from file menu

existing Projects Into Workspace

select folder LCP176x

put the flag on Copy projects into workspace

click finish

am I wrong something?

terjeio, 2021-07-26:
Does your workspace look similar to the one I posted above?

silvanopas, 2021-07-27:
It is identical

terjeio, 2021-07-27:
Ok, what about include paths? Are they the correct? Use _File > Properties_ and navigate to this dialog and compare:

[image]

The [.cproject](https://github.com/grblHAL/LPC176x/blob/master/.cproject) file in the project folder contains settings such as these - I wonder if it has somehow been corrupted.

silvanopas, 2021-07-27:
Good news.
I have reinstalled MCUXpress ide and now it compiles without errors.
Most likely I had some corrupted files ....

Thanks and sorry if I wasted your time.
As soon as I can I update you on the card tests

silvanopas, 2021-07-28:
Hello
I was able to compile and generate the firmware.bin file to be installed on
the skr 1.4 turbo.
Obviously I have configured the card in the my_machine.h file. (#define
BOARD_BTT_SKR_14_TURBO).
However, once the firmware is installed, the serial port is not recognized.
Do I need to configure anything else?
Where can I define the baudrates.
Thank you

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mail
priva di virus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>