grblHAL/LPC176x #7: warnings on compile

Cncninja, 2021-09-06:
Are this anything to worry about?

[image]

[image]

terjeio, 2021-09-07:
No, I just ignore these warnings. It is due to some bad code from NXP, perhaps one should take a look at why they are issued...

Cncninja, 2021-09-20:
The 2130 still has errors with a fresh download as well as the SD card plug-in. i deleted the 2130 and re-commented out the SD card and it compiled. I don't use 2130s or the SD, so no issue for me but just letting you know in case the plugins are affecting other builds. 
Thank you, do you have a donation link?

terjeio, 2021-09-20:
> The 2130 still has errors 

Which errors? It compiles without errors for me with both 2130 and SD card enabled.
The iap warnings are on NXP - not my code.

> do you have a donation link?

Sorry, I do not.

Cncninja, 2021-09-21:
[image]
[image]
I deleted the 2130 folders so i don't see those errors but this is the SD card error, I don't use the SD so no problem for me but not sure if this would affect other builds.
Thank you for your work.

terjeio, 2021-09-21:
Ok, if you uncomment SDCARD_ENABLE you have to [fetch the FatFs source](https://github.com/grblHAL/LPC176x/tree/master/src/FatFs), I'll add it to the next commit so that downloading will no longer be required .