grblHAL/LPC176x #8: io stream error with new update push cant compile

Cncninja, 2021-09-12:
[image]

What should i do?

terjeio, 2021-09-12:
> What should i do?

I think updating submodules will fix it, info [here](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL#using-git-command-line).

Cncninja, 2021-09-13:
I have fetched the latest using git and tried with Github desktop as well shows the latest from 3 days ago. both ways I now get 358 errors im not changing anything but changing from smoothie to SKR1.4 and 5150 support.
[image]

Cncninja, 2021-09-13:
[image]

Cncninja, 2021-09-13:
Thanks for the help it seems all related to the new trinamic folder.. Again thank you for the help

terjeio, 2021-09-13:
The Trinamic library has somehow been corrupted when I last updated the repo. I've just commited a fix (for this driver only) that should compile.

Perhaps this fixes your [other issue](https://github.com/grblHAL/Plugins_motor/issues/1#issuecomment-917751249) as well? Please report back.

Cncninja, 2021-09-13:
Seem most errors went away, but still left with 25 from the 2130.c folder. full new fresh download
[image]

terjeio, 2021-09-13:
More git trouble - just delete trinamic2130.c and trinamic2130.h for now. I'll commit a fix that does that later.

terjeio, 2021-09-15:
Fix commited, reopen if still an issue.

closed 2021-09-15 by terjeio