grblHAL/LPC176x #9: $22 it does not affect homming

silvanopas, 2021-09-14:
Hi
I can't activate homing, I put $ 22 = 1 but nothing changes, am I doing something wrong?

terjeio, 2021-09-14:
What are the values for $44, $45 and $46? These are the axis masks for which axes to home during pass 1 to 3.

If all are 0 nothing will happen on a $H command. BTW do you get an error response when homing?

silvanopas, 2021-09-15:
$22=1
$44=0 , $45=0, $46=0
With $H
Lightburn replies: Homing cycle is not enabled in your Grbl settings.
Lasergrbl setting disabled

terjeio, 2021-09-15:
Ok, if you want to home Z first then X and Y you have to set $44=4 and $45=3.

More info [here](https://github.com/grblHAL/core/wiki/Additional-or-extended-settings).

silvanopas, 2021-09-15:
Now works
where can i find information on these parameters? ($44 $45 $46)

thanks

terjeio, 2021-10-11:
> where can i find information on these parameters? ($44 $45 $46)

One option is to use inbuilt help from a terminal:

`$help homing` lists all homing related settings, with a description in the latest build.

`$$=44` for info about setting 44...

[This wiki page](https://github.com/grblHAL/core/wiki/Additional-or-extended-settings) has information about new settings as well.

closed 2022-01-21 by terjeio