grblHAL/LPC176x #10: how to modify PINs ?

drom89, 2021-10-03:
Hi again, just a dumb question, how can i modify pin outputs ? 

According to PIN map for SKR 1.4T [https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4PIN.pdf]
I suppose that it should be done in btt_skr_1.4_turbo_map.h ?
I want to use PIN 2.7 for PWM output to CO2 laser PSU.
Because i have one SKR 1.4T board where all other MOSFETs are fried :-)

And it is possible to control outputs thru G code ?

terjeio, 2021-10-04:
> Hi again, just a dumb question, how can i modify pin outputs ?

By modifying the pin map file or creating a new.

> I suppose that it should be done in btt_skr_1.4_turbo_map.h ?

Yes, but better to copy  it to _my_machine_map.h_ and enable that in _my_machine.h_ instead. This way your changes will not be overwritten on an update.

> I want to use PIN 2.7 for PWM output to CO2 laser PSU.

According to the datasheet this pin is not PWM output capable, so not possible. Pins supported by code can be found [here](https://github.com/grblHAL/LPC176x/blob/911c8ee9dac161b89385105ca2c5736dc1b7f6d2/src/grbl-lpc/pwm_driver.c#L49-L95).

> And it is possible to control outputs thru G code ?

Yes, currently up to 3 can be controlled with M62 - M65. Pin assignments has to be added to the map file, an example:

https://github.com/grblHAL/LPC176x/blob/911c8ee9dac161b89385105ca2c5736dc1b7f6d2/src/ramps_1.6_map.h#L189-L197

drom89, 2021-10-04:
> > I want to use PIN 2.7 for PWM output to CO2 laser PSU.
> 
> According to the datasheet this pin is not PWM output capable, so not possible. Pins supported by code can be found [here](https://github.com/grblHAL/LPC176x/blob/911c8ee9dac161b89385105ca2c5736dc1b7f6d2/src/grbl-lpc/pwm_driver.c#L49-L95).
> 
What datasheet are you looking at ? 

On SKR1.4T  PIN2.7 is connected to mosfet and mosfet is controlled from Pin 66 on the chip if im correct.
BTT have it market as  PWM but in data sheet for MCU its not labeled as PWM hmm, im getting lost in this.
But i think, that it can be used as PWM, i will try test it.

[image]
[image]
[image]

terjeio, 2021-10-04:
> What datasheet are you looking at ?

The MCU user manual - UM10360, same info there as in the last image in the comment above.

> But i think, that it can be used as PWM, i will try test it.

All GPIO pins can be used for PWM if you use timer(s) to generate interrupt(s) and flip the output in the interrupt handler(s).  With a 5KHz PWM frequency that will be two interrupts firing every 200 microseconds, possibly very close - you can get away with that without causing too much jitter?

drom89, 2021-10-06:
Hi, i have some small problems.

i want to use pin P2.3 (PWM1_CH4 / pin70 on the chip) on SKR to #define SPINDLE_ENABLE_PIN      23

when i try #define SPINDLE_ENABLE_PIN PWM1_CH4 it doesnt work
so i tried #define SPINDLE_ENABLE_PIN 70 it doesnt work 

I want it cuz i got some weird stuff going on with laser, when using default 23 its always on at least little bit. 
What im doing wrong ?

terjeio, 2021-10-12:
Fast direction of engraving is horizontal in the images above? If so it is strange yes.

Are your belts and pulley setscrews tight?
And was the STOCK K40 engraving with the same mechanical configuration and not before any modifications?

drom89, 2021-10-19:
So it looks like its mechanical thing.

drom89, 2021-11-25:
So $35 helped, didnt test how fast i can go.

closed 2022-01-21 by terjeio