grblHAL/LPC176x #12: PlatformIO/Makefile compilation

dzervas, 2021-11-10:
I've been trying to compile this project without installing a huge IDE that I need once-twice a year and Arduino reports "no libraries found".

Can you please give the full compiler flags, declarations and linker options that you have so I can write some kind of a nicer build method? Maybe even a GitHub Action template that people can fork and maintain their builds nicely.

I came up with this `platformio.ini`:
```
[platformio]
src_dir = src
lib_dir = src

[env:LPC176x]
platform          = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip
platform_packages = framework-arduino-lpc176x@^0.2.8
framework         = arduino
board             = nxp_lpc1769
; lib_ldf_mode      = chain+
lib_ldf_mode      = off
lib_compat_mode   = strict
lib_archive = yes

src_filter = +<./*/*>
;            =-<./src/eeprom>
build_flags =
    ; Include paths
    ; -Isrc/eeprom
    -Isrc
    -Isrc/FatFs
    -Isrc/grbl
    ; -Isrc/grbl-lpc
    ; -Isrc/lpc_chip_175x_6x
	-Isrc/lpc_chip_175x_6x/inc
	-Isrc/lpc_chip_175x_6x/inc/usbd
    ; -Isrc/lpc17xx

    ; Compiler options
    -D__CODE_RED
    -DCORE_M3
    -D __USE_CMSIS=CMSIS_CORE_LPC17xx
    -D__LPC17XX__
    -D__NEWLIB__
    -D__USE_LPCOPEN
    -DDEBUG

	; -fpermissive
    -c
    -fmessage-length=0
    ; -fno-builtin
    -ffunction-sections
    -fdata-sections
    -fno-common
    ; -nostdinc
    -lusbd_175x_6x_lib
    -Lsrc/lpc_chip_175x_6x/libs

    ; Machine Options
    -DNO_BOARD_LIB
```

The lines starting with `;` are comments (due to rigorous testing)

The error I get is the following:
```
In file included from /home/dzervas/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/include/debug_frmwrk.h:35,
                 from /home/dzervas/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/lib/usb/cdcuser.h:22,
                 from /home/dzervas/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/lib/usb/usbcore.cpp:39:
/home/dzervas/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/include/lpc17xx_uart.h:598:20: error: 'TRANSFER_BLOCK_Type' has not been declared
  598 |   uint32_t buflen, TRANSFER_BLOCK_Type flag);
      |                    ^~~~~~~~~~~~~~~~~~~
/home/dzervas/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/include/lpc17xx_uart.h:600:20: error: 'TRANSFER_BLOCK_Type' has not been declared
  600 |   uint32_t buflen, TRANSFER_BLOCK_Type flag);
      |                    ^~~~~~~~~~~~~~~~~~~
Compiling .pio/build/LPC176x/CMSIS/lib/usb/usbuser.cpp.o
*** [.pio/build/LPC176x/CMSIS/lib/usb/usbcore.cpp.o] Error 1
Compiling .pio/build/LPC176x/CMSIS/system/startup_LPC17xx.S.o
In file included from /home/dzervas/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/include/debug_frmwrk.h:35,
                 from /home/dzervas/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/lib/usb/usbuser.cpp:21:
/home/dzervas/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/include/lpc17xx_uart.h:598:20: error: 'TRANSFER_BLOCK_Type' has not been declared
  598 |   uint32_t buflen, TRANSFER_BLOCK_Type flag);
      |                    ^~~~~~~~~~~~~~~~~~~
/home/dzervas/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/include/lpc17xx_uart.h:600:20: error: 'TRANSFER_BLOCK_Type' has not been declared
  600 |   uint32_t buflen, TRANSFER_BLOCK_Type flag);
      |                    ^~~~~~~~~~~~~~~~~~~
*** [.pio/build/LPC176x/CMSIS/lib/usb/usbuser.cpp.o] Error 1
```

What toolchain and framework are you using? Any specific versions?

terjeio, 2021-11-11:
MCUXpresso IDE v11.3.1_5262,  gcc. Framework by NXP from 2014 - included in the download.

Compiler command line:

`-DDEBUG -DNO_BOARD_LIB -DCORE_M3 -D__LPC17XX__ -D__CODE_RED -D__NEWLIB__ -D__USE_LPCOPEN -I"D:\Projects\NXP\GRBL Driver LPC176x\src" -I"D:\Projects\NXP\GRBL Driver LPC176x\src\FatFs" -I"D:\Projects\NXP\GRBL Driver LPC176x\src\lpc_chip_175x_6x\inc" -I"D:\Projects\NXP\GRBL Driver LPC176x\src\lpc_chip_175x_6x\inc\usbd" -O0 -fno-common -g3 -Wall -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -fmerge-constants -fmacro-prefix-map="../$(@D)/"=. -mcpu=cortex-m3 -mthumb -D__NEWLIB__ -fstack-usage`

Linker:

`-nostdlib -L"D:\Projects\NXP\GRBL Driver LPC176x\src\lpc_chip_175x_6x\libs" -Xlinker --defsym=__heap_size__=0xC00 -Xlinker -Map="GRBL Driver LPC176x.map" -Xlinker --cref -Xlinker --gc-sections -Xlinker -print-memory-usage -mcpu=cortex-m3 -mthumb -T "GRBL_Driver_LPC176x_Debug.ld"`

`TRANSFER_BLOCK_Type` is a Arduino specific type, it is not defined in the NXP framework. A framework bug?

dzervas, 2021-11-17:
I've sadly resorted to MCUXpress... :disappointed: 
It requires the NXP SDK which is not an easy task to find, setup, make the compiler work and find everything...

Closing this as "too much pain" for now

closed 2021-11-17 by dzervas

terjeio, 2021-11-18:
I do not understand that you have to find the NXP SDK, is it for generating a makefile?
The SDK files is contained in the download.

dzervas, 2021-11-18:
oh I meant setting up the NXP SDK without MCUXpress. Compiling using MCUXpress went smoothly by changing a backslash to slash in https://github.com/grblHAL/LPC176x/blob/master/src/main.c#L24

terjeio, 2021-11-18:
> oh I meant setting up the NXP SDK without MCUXpress.

Yes, I understood that - PiO cannot use the SDK files present in the download?

I'll fix the backlash in the next commit - Windows masks these "bugs".

dzervas, 2021-11-18:
I'd need to create a new framework for PiO that's just the NXP SDK - I'm not into that (yet - at least I wanna see my CNC cutting)