grblHAL/LPC176x #13: TMC2209 on SKR E3 Turbo report incompatible driver

dzervas, 2021-11-17:
I'm trying to get E3 Turbo to work, which has soldered TMC2209s but after writing a map and setting `TRINAMIC_ENABLE 2209` I get via serial:

```
error:7
grblHAL: incompatible driver
```
The weird thing is that for a moment it worked but somehow didn't compile the trinamic code (no `$338` setting available).

`src/my_machine.h`

```c
/*
  my_machine.h - configuration for NXP LPC176x ARM processors

  Part of grblHAL

  Copyright (c) 2020-2021 Terje Io

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

// NOTE: Only one board may be enabled!
// If none is enabled pin mappings from generic_map.h will be used
//#define SMOOTHIEBOARD
//#define BOARD_RAMPS_16
//#define BOARD_BTT_SKR_13
//#define BOARD_BTT_SKR_14_TURBO
#define BOARD_BTT_SKR_E3_TURBO
//#define BOARD_MKS_SBASE_13
//#define BOARD_MY_MACHINE // Add my_machine_map.h before enabling this!

// Configuration
// Uncomment to enable, for some a value > 1 may be assigned, if so the default value is shown.

#define USB_SERIAL_CDC     1 // Comment out to use UART communication.
#define SDCARD_ENABLE      1 // Run gcode programs from SD card, requires sdcard plugin.
//#define TRINAMIC_ENABLE 2209 // Uncomment to enable Trinamic TMC2130 driver support. NOTE: Experimental for now, currently for SKR 1.x boards only
//#define TRINAMIC_ENABLE 2130 // Uncomment to enable Trinamic TMC2130 driver support. NOTE: Experimental for now, currently for SKR 1.x boards only
//#define TRINAMIC_ENABLE 5160 // Uncomment to enable Trinamic TMC5160 driver support. NOTE: Experimental for now, currently for SKR 1.x boards only
//#define LIMIT_MAX_ENABLE   1 // Uncomment to enable max limit input pins (when available)
#define EEPROM_ENABLE      1 // I2C EEPROM support. Set to 1 for 24LC16(2K), 2 (64 byte pages) or 3 (32 byte pages) for larger sizes. Requires eeprom plugin.
//#define EEPROM_IS_FRAM     1 // Uncomment when EEPROM is enabled and chip is FRAM, this to remove write delay.

// If the selected board map supports more than three motors ganging and/or auto-squaring
// of axes can be enabled here.
#define X_GANGED            1
#define X_AUTO_SQUARE       1
#define Y_GANGED            1
#define Y_AUTO_SQUARE       1
//#define Z_GANGED            1
//#define Z_AUTO_SQUARE       1
// For ganged axes the limit switch input (if available) can be configured to act as a max travel limit switch.
// NOTE: If board map already has max limit inputs defined this configuration will be ignored.
//#define X_GANGED_LIM_MAX    1
//#define Y_GANGED_LIM_MAX    1
//#define Z_GANGED_LIM_MAX    1

/*EOF*/
```
[... 198 lines cut ...]

#endif

/**/
```

`src/driver.h` change to include my config

```diff
diff --git a/src/driver.h b/src/driver.h
index 20551f5..084cdce 100644
--- a/src/driver.h
+++ b/src/driver.h
@@ -81,6 +81,8 @@
   #include "btt_skr_1.3_map.h"
 #elif defined(BOARD_BTT_SKR_14_TURBO)
   #include "btt_skr_1.4_turbo_map.h"
+#elif defined(BOARD_BTT_SKR_E3_TURBO)
+  #include "btt_skr_e3_turbo_map.h"
 #elif defined(BOARD_MKS_SBASE_13)
   #include "mks_sbase_map.h"
 #elif defined(BOARD_MY_MACHINE)
```

Ideas? Maybe the EEPROM is corrupted? Can I reset it via an SD card file for example?

terjeio, 2021-11-18:
There are some checks in [grbl/grbllib.c](https://github.com/grblHAL/core/blob/0c47a688b57593b77fad629add4e7dcf958d49cf/grbllib.c#L141-L192) to verify that the driver is usable. Settings and HAL version number match is basic and done in the driver and returned to grbllib as a bool, these are correct. Checks for variable spindle support and safety door input is among those done against driver capabilities flags and I would guess one of these fails.

So first thing to do is to check that ENABLE_SAFETY_DOOR_INPUT_PIN is not uncommented in _grbl/config.h_ as your pin map file does not have a mapping for that.

terjeio, 2021-11-18:
I have added a reason code (a bitfield containing flags) to the incompatible driver message, replace _grbllib.c_ with this version:

[grbllib.zip](https://github.com/grblHAL/LPC176x/files/7561360/grbllib.zip)

Note that this requires a recent version of the core, perhaps the latest, to compile.

dzervas, 2021-11-18:
BTW shouldn't the core and the rest of the libs be git submodules?

thank you very much for your response, will test everything later tonight

dzervas, 2021-11-18:
What I also don't get about the 2209s is how to set them up in the map. They need CS, right? Otherwise you need a specific UART port for each driver or software UART - which both sound stupid in my head.
From what I see in the "docs" for the board (more like a 5 picture PDF) the "UART pin" is the CS pin and it's not on the same PN for every driver:
[image]
link to manual: https://www.roboter-bausatz.de/media/pdf/03/15/06/BTT-SKR-E3-Turbo-User-Manual.pdf

How would I setup the `MOTOR_CS<axis>_PIN`?

terjeio, 2021-11-18:
> BTW shouldn't the core and the rest of the libs be git submodules?

They are, except the NXP SDK which is part of the driver repo.

> What I also don't get about the 2209s is how to set them up in the map. They need CS, right?

No - they do not support SPI mode.

> Otherwise you need a specific UART port for each driver or software UART - which both sound stupid in my head.

Yes - kind of stupid to have a software UART for each as up to four can be controlled by adressing the driver in the messages sent. So only two hardware UARTs are needed used for 5 axes.

This driver does not have support for software UARTs, that has to be added.
@fitch22 did that for the STM32F4xx driver, code is [here](https://github.com/grblHAL/STM32F4xx/blob/master/Src/btt_skr_2.0.c).

dzervas, 2021-11-18:
huh, so there's currnetly no support for this board due to the UART problem? :disappointed:

dzervas, 2021-11-18:
Another problem is that I can't seem to get working is save settings to EEPROM. For some reason, every time 
I hard reset the board (via the button) all my settings are lost...

dzervas, 2021-11-18:
ooooh cool cool - that makes everything so much clearer! thank you!

I'll start porting the 2209 code from the STM32 driver

dzervas, 2021-11-19:
I'm having a hard time translating the interrupt timer/counter code from the STM32 variant:

```c
		MX_TIM7_Init();
		htim7.Instance->CR1 |= TIM_CR1_UDIS;			// Disable update events
		HAL_TIM_Base_Start_IT(&htim7);
```

and the whole `MX_TIM7_Init` function. Can you nudge me in the right direction? I'm having quite a hard time figuring out how the internal clocks of the LPC work - the rest of the code is almost the same and seems to be ok

fitch22, 2021-11-19:
Dimitris,

The gist of how this works is there is a timer used for an interrupt, the rate is the bit rate for the serial protocol.  The timer is normally disabled so that interrupts do not happen unless you are transmitting or receiving.  For the STM32 version I have, Timer 7 is the least capable of all the timers and thus the simplest as well.

 

The code snippet you show initializes the counter, then sets the update disable bit (this stops the rollover of the counter from generating an interrupt) and then sets up the timer to generate an interrupt.  The code is all part of what is provided by ST and will likely have something similar provided by NXP.

 

For a serial transmission, the bits are output 1 bit every interrupt.  At the end of a write (no readback) the interrupts are turned off.

 

For a serial reception, first the read request is sent to the 2209 similar to a write, but after the write the output pin is switched to an input and the pin is polled for the beginning of the START bit (low).  When the transition is found at the beginning of the START bit, the counter used for the timer is adjusted so that interrupts will occur in approximately the center of each bit period.  The number of bits to be received are captured after which the interrupts are again turned off and the pin is changed back to an output.

 

I am not familiar with the timers or interrupts in the LPC processors, but there is most likely something very similar.  I have a SKR 1.4 Turbo if you need me to test anything when you are ready.

 

Bill

terjeio, 2021-11-19:
It is a long time since I worked with peripherals on this MCU, and I like to access registers directly bypassing library code. If this does not put you off then take a look at how the main stepper and step pulse timers are set up and handled in [driver.c](https://github.com/grblHAL/LPC176x/blob/master/src/driver.c), init code is [here](https://github.com/grblHAL/LPC176x/blob/79acd36ce76bfe71663532746906e2a60e132359/src/driver.c#L1227-L1248).

You can also try to dig into the [Marlin source](https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin/src) to see how they have implemented soft UARTs.

dzervas, 2021-11-19:
woah, way, way more info than I was expecting! I think that I have enough to start digging! Thanks a TON!!!!

dzervas, 2021-11-19:
Hmmm, I've done a first try here: https://github.com/dzervas/LPC176x/blob/master/src/btt_skr_2.x.c and when I send `$338=1` or `$338=7` the controller just crashes

terjeio, 2021-11-20:
```
void SUART_IRQHandler (void)
{
    sprintf(sbuf, "IRQHandler Fired" ASCII_EOL);
    hal.stream.write(sbuf);

	if (tx_busy)
		send();
	if (rx_busy) {
		rcv();
		rx_irq_count++;
	}
}
```

Calling sprintf and writing a long string in the handler is not a good idea, sprintf is slow and writing may fill the output buffer causing it to block the system? Write a single character or toggle a pin is better.
BTW calling sprintf is is just overhead. You can use `hal.stream.write("IRQHandler Fired" ASCII_EOL);` to write a constant string.

The reason for the crash could be that you do not clear the interrupt flag in the handler? The first line in the stepper timer handler does that, writing a 1 to a bit in the IR register clears it - to clear all write the current value back:

```
// Main stepper driver
void STEPPER_IRQHandler (void)
{
    STEPPER_TIMER->IR = STEPPER_TIMER->IR;
    hal.stepper.interrupt_callback();
}

```

From the manual:

> The Interrupt Register consists of 4 bits for the match interrupts and 4 bits for the capture
interrupts. If an interrupt is generated then the corresponding bit in the IR will be high.
Otherwise, the bit will be low. Writing a logic one to the corresponding IR bit will reset the
interrupt. Writing a zero has no effect. 

`NVIC_SetPriority(SUART_TIMER_INT0, 2); // Same with stepper timer - no idea why`

Lower value is higher priority, higher proirity interrups will be handled first and will also preempt the handlers of those with lower priority. BTW if you print anything the priority should be lower (higher value) than the interrupt(s) involved in outputting the data.

dzervas, 2021-11-20:
Yea I gave 0 thought about `sprintf` - I just wanted to see letters on my terminal :stuck_out_tongue: 
The buffer doesn't seem to fill - I'd get at least some letters of the buffer but all I get is this after sending `$338=1`:

```
if_init
if_init post
UART read
writen
writen2
```

I added the code to clear the IR but no luck, same behavior. I wanna try to use the higher layer timer functions `TIMER_Init`, `TIMER_AttachInterrupt`, etc. Might not be your cup of tea but at least I wanna see the IRQ firing and then I can clean the code :pray: (and remove all those horrible "debugging" printfs)

terjeio, 2021-11-20:
> They are not available (?)

Seems like it - I am using bits of the `lpc_chip_175x_6x` API that is basically wrapper functions for register access.
Switching to a higher level API will cause a lot of work in other parts of the code?

BTW I cannot find where you set the timeout period, this is how it is done for the stepper timer:

```
    STEPPER_TIMER->TCR = 0b10;          // reset
    STEPPER_TIMER->MR[0] = 0xFFFF;      // Set a long initial delay,
    STEPPER_TIMER->TCR = 0b01;          // start stepper ISR timer in up mode
```

You have to set a delay in `MR[0]` that gives you the baud rate you are after.

dzervas, 2021-11-20:
THAT WAS IT! MY TYRMINAL IS FULL WITH 'a'!!!!!!!! YAY!
Ok of course my timings are waaaaay off but now it's "just" timing math :tada: 

You da best!

dzervas, 2021-11-21:
I can't seem to drive a signal. It's as if `DIGITAL_IN`/`DIGITAL_OUT` dont work. I'm guessing I'm doing something wrong with `PinMuxSet` and the `FUNC0` which I can't wrap my head around.

I've temporarily mapped the Z stepper CS (UART pin) to an accessible pin and I've attached a chinese logic analyzer that I have hanging around and the signal remains high no matter what I do. I've tried both pullup and pulldown, using PinMuxSet, PinMux... No luck.

Do you see any obvious problems with how I drive the lanes?

terjeio, 2021-11-21:
Pinmuxing is done for you [here](https://github.com/dzervas/LPC176x/blob/42fdae0d00ca4ea78ba6a90c37967d0b7971cf00/src/driver.c#L1209-L1223) where the pins are set as output. You do not need to do that again.

I guess you have been mislead by the STM32 code where the pin number is used to interact with the port (via bitband access), this MCU uses a bit mask since it does not support bitbanding. So change:
`cs[X_AXIS].bit = gpio->pin;`
to:
`cs[X_AXIS].bit = gpio->bit;`

Hopefully this helps...

dzervas, 2021-11-21:
Well I toggle between input & output so I gotta set the pinmuxing again (right?)

What you said was right and also I had `bit` as `uint16_t` so it was overflowing -.- (how I miss Rust...)

terjeio, 2021-11-21:
> Well I toggle between input & output so I gotta set the pinmuxing again (right?)

No, I do not think so - flipping the correct DIR bit should be enough. Pinmuxing is for routing the pin to the GPIO peripheral.

dzervas, 2021-11-22:
After much debugging I'm able to communicate with some drivers! (side note due to each driver being in a different UART lane, all of them have slave address 0!)

Problem is that I'm getting communication error with E0 (X1) and Z drivers, which is very odd. I'm not able to swap drivers as they're soldered on. Has this happened to you before? ideas?

dzervas, 2021-11-22:
It was the standby pins! I added support for them and pulling them down on startup :tada: 
I'll be preparing a PR soon

terjeio, 2021-11-22:
Great that you have progress!

Hmm, the slave address beeing 0 for all has to be somehow fixed somehow?
Do you set it to 0 [here](https://github.com/terjeio/Trinamic-library/blob/2b73364356b02918543270af8e1407bee83d3654/tmc2209hal.c#L326) for now?

dzervas, 2021-11-22:
I've added a compile flag to hardcode the address for all motors. It's not perfect (you can't set a custom address for each motor) but it's good enough

terjeio, 2021-11-22:
> I've added a compile flag to hardcode the address for all motors.

I am not too keen on that - I would rather like the low level driver code to report back the address to use.

dzervas, 2021-11-22:
Oh cool, I'll transfer the slave address initialization somewhere in driver.c? Do you have any better ideas?

BTW I see that the TMC mode is hardcoded. Do I have to change it to enable stallguard and sensorless homing (I know that there's a runtime setting for sensorless but I think it depends on the mode that the driver is to enable it? No?)

terjeio, 2021-11-22:
> Do you have any better ideas?

Yes - I want the addresses transferred from the low level driver code (your code) as this may be board dependent.

> BTW I see that the TMC mode is hardcoded.  Do I have to change it to enable stallguard and sensorless homing...

No, many of the hardcoded values are just defaults. The trinamic folder contains code that can be used in other projects than grblHAL, in fact it is located in my private repo. _trinamic.c_ in the motors folder is the "middleware" between grblHAL and the chip specific driver code. The low level driver code is the glue that binds everything together.

dzervas, 2021-11-22:
> Yes - I want the addresses transferred from the low level driver code (your code) as this may be board dependent.

Cool! Shouldn't be too hard

> No, many of the hardcoded values are just defaults

So how would I enable stallguard? I don't see any other config or runtime setting

terjeio, 2021-11-22:
> So how would I enable stallguard? I don't see any other config or runtime setting

Some info [here](https://github.com/grblHAL/Plugins_motor).

dzervas, 2021-11-22:
god I'm blind...

Thank you very very much for you patience and help!!!

terjeio, 2021-11-22:
To be honest sensorless homing is complicated to get right. There are parameters to tune in driver.h as well such as [these](https://github.com/grblHAL/Plugins_motor/blob/913ab1561fa6094e48893742882938f280b9c5da/trinamic.h#L70-L71), and a ton of info in the Trinamic chip documentation. IMO easier/better to use limit switches...

dzervas, 2021-11-22:
From discussions with friends for the past hour it seems that nobody likes sensorless homing - but somehow my mind is set that it's the most repeatable solution. Of course I have to tinker with it a lot (maybe even write some code to make runtime configuration available if some variables are missing) as the cutting forces are much greater than the 3D printing almost negligent forces...

terjeio, 2021-11-22:
I have added a callback function that will let the low level driver set the slave address. It will be called for each enabled motor.

```
void motor_preinit (motor_map_t motor, uint8_t *address)
{
    *address = 0;
}

void board_init(void)
{
    static trinamic_driver_if_t driver_if = {
        .on_drivers_init = if_init,
        .on_motor_preinit = motor_preinit
    };

    trinamic_if_init(&driver_if);
}
```

This will not break any existing code, do you want to try it? Note that I may change the signature of this call before a final commit. If you want to try I'll add a zip to a comment with the changes.

> (maybe even write some code to make runtime configuration available...)

FYI [ioSender](https://github.com/terjeio/ioSender) has a Trinamic tuning tab for the basic stuff... Have you seen [this discussion](https://github.com/grblHAL/core/discussions/41)?

terjeio, 2021-11-22:
> (also I need camera alignment feature which the ioSender lacks)

Is that something else than this? 😉

[image]

Which sender are you using?

> I'll try the callback!

I'll prepare a zip after dinner.

dzervas, 2021-11-22:
oh cool! Linux support is still a no-go though :disappointed: 

> Which sender are you using?

bCNC

There seems to be a bug and I can't control X2 & Y2 through M122. Should I open an issue in the trinamic repo?

terjeio, 2021-11-22:
I got  a visitor when I was about to post the zip so I was a bit distracted - I do not know if I posted it on the wrong issue or if it has been lost in cyberspace...

Anyway here it is:

[motors.zip](https://github.com/grblHAL/LPC176x/files/7583587/motors.zip)

I have not tested it, only that it compiles.

> There seems to be a bug and I can't control X2 & Y2 through M122. Should I open an issue in the trinamic repo?

Yep, I think that is the best.

terjeio, 2021-11-28:
New version out now with some bug fixes and final API change for setting driver slave id. Example [here](https://github.com/grblHAL/STM32F4xx/blob/014624242955feb3965b6a052e2aceb95d071ef9/Src/btt_skr_1.1.c#L637-L650). I have also added UART specific pin id's, no need to hijack the chip select id's anymore - you may check the rest of the example and the corresponding [map file](https://github.com/grblHAL/STM32F4xx/blob/014624242955feb3965b6a052e2aceb95d071ef9/Inc/btt_skr_pro_v1_1_map.h#L151-L203) for details.

dzervas, 2021-11-29:
That makes the whole setup much easier!

I hope I'll be able to test it soon (got caught up with the rest of the machine)
The only other thing that's missing in the API level is the standby pin for some trinamic drivers (my board has them and needs them pulled low as early as possible)

dzervas, 2021-12-01:
I haven't tested any of the code yet as it seems that currently my code "works" and I'm having a problem just with the X0 driver. The motor does not move normally CW/CCW it just rattles for the length of the movement. All the other motors move as expected.

When I switched X0 with X1 motor, it was moving flawlessly and the X1 motor (with the X0 driver) had the exact same issue. Do you have any idea what that could be? My drivers are soldered on so I can't swap them

I really hope I won't have to ditch the board just because a single driver went crazy...

terjeio, 2021-12-02:
Bad driver?
Since you are using Linux you may try to [enable live reporting](https://github.com/grblHAL/Plugins_motor#m122---output-debug-info-or-reset-driver) of the stallguard result, for that you have to enable A and B axes (`N_AXIS = 5`) in _grbl/config.h_ instead of ganged motors. That will also isolate any potential code issues for ganged axes. `M122AS1` followed by motion commands output a stream of values from the A (X0) driver. Compare that to what is output from working drivers. If you mechanically load the motors the reported value should change.

dzervas, 2021-12-11:
I'm through with this horrible piece of a board... It's crap and even the schematic provided by the manufacturer is wrong. I've changed to an ESP32 & FluidNC board

Thank you very very much @terjeio, you were extremely helpful and supportive. Keep it up! :)
I'm attaching my work as of now which seems to work good enough but needs some cleanup to be MR-able
[Uploading GRBLHal-LPC176x.zip…]()

closed 2021-12-11 by dzervas

terjeio, 2021-12-11:
I am sorry to hear this, anyway I have modified your code a bit to make it self-contained and I will add it to the next release so your work is not lost.

Thank you for your effort.