grblHAL/LPC176x #14: Laser head is slowing down during rastering

drom89, 2021-11-26:
Hi,

another "problem" when engraving/rastering with higher speeds the laser head is slowing down considerably, is this normal ?
I thought that laser head should be running with consistent speed and only change the power of the laser.

See video below
[https://photos.app.goo.gl/U3dMczxaXWyc69ma8](url)

terjeio, 2021-11-26:
Yes, it is normal because lots of short movements causes the planner buffer to limit the feed rate. Try with a larger [planner buffer](https://github.com/grblHAL/core/blob/fc0f9ecf75b74e6e5f39fcd90cd6c54fe3000cb2/planner.h#L27-L30). Each entry requires around 40 bytes of RAM.

FYI the iMXRT1062 driver, which has a 600MHz processor with FPU and plenty of RAM, can be tuned to feedrates > 50.000 mm/min for laser engraving. 

Some info [here](https://github.com/terjeio/grblHAL/discussions/162) and [here](https://github.com/terjeio/grblHAL/issues/142).

drom89, 2021-12-03:
So i boosted the planner buffer from 36 to 180.
Im using lightburn and with synchronous transfer mode its still almost same.
With buffered mode its ok but only like 10 sec and after that laser head stop moving at all and youst fire in place.

I think USB communication is too slow... 
Is there any way to boost USB baud rate from default 115200 to higher ?

terjeio, 2021-12-03:
zip and attach a file that shows the symptom to acomment so I can check here.

>Is there any way to boost USB baud rate from default 115200 to higher ?

Which board map are you using?

drom89, 2021-12-03:
SKR 1.4 Turbo board.
Tomorrow i will record video.

drom89, 2021-12-03:
When trying to go with 200 buffer size i run our of memory.
[image]

terjeio, 2021-12-03:
> Tomorrow i will record video.

I do not need one, I would rather have a file showing the symptoms so I have something to test with.

> When trying to go with 200 buffer size i run our of memory.

Not much I can do about that, it is an old processor with limited RAM. I do not know if the AHB region can be made accessible by tweaking the linker script, you can look into that? And perhaps it is not something that will gain you any speed - there is no FPU (floating point unit) in this processor so too many entries in the buffer may start to slow it down instead.

Can you run a program in Check mode and time that? That will give you the raw transfer speed. With a native USB connection (IIRC the SKR 1.4 uses that) the configured speed does not matter, AFAIK the max speed available will be used anyway. At least it is for the iMXRT1062 driver.

drom89, 2021-12-03:
> > Tomorrow i will record video.
> 
> I do not need one, I would rather have a file showing the symptoms so I have something to test with.

Do you want G code file ? 

> Not much I can do about that, it is an old processor with limited RAM. I do not know if the AHB region can be made accessible by tweaking the linker script, you can look into that? And perhaps it is [...]

What boards do you recommend performance/cost wise ? For lasers, cnc mills etc. 

> Can you run a program in Check mode and time that? That will give you the raw transfer speed. With a native USB connection (IIRC the SKR 1.4 uses that) the configured speed does not matter, AFAIK the max speed available will be used anyway. At least it is for the iMXRT1062 driver.

How can i run it in Check mode ?

terjeio, 2021-12-03:
> Do you want G code file ?

Yes please.

> What boards do you recommend performance/cost wise ? For lasers, cnc mills etc.

That is a hard one to answer - the iMXRT1062 (Teensy 4) driver and board is the fastest and most expandable you can get but needs external drivers as there is currently no boards available for Polulu style drivers. A processor with FPU is IMO preferable, but the Pi Pico is interesting due to its PIO peripherals. I have not checked what its limits are though. More info [here](https://github.com/grblHAL/Controllers). BTT SKR Pro 1.1/1.2 boards will hopefully soon be programmable via SD card - I had a breaktrough yesterday with a dev board that brings that closer.

> How can i run it in Check mode ?

By sending` $C` before starting. Some senders has a button or a checkbox for enabling Check mode.

drom89, 2021-12-03:
Ok thanks, i will test it tomorrow and send the files.

drom89, 2021-12-05:
files are here: [gcode foto.zip](https://github.com/grblHAL/LPC176x/files/7654839/gcode.foto.zip)

terjeio, 2021-12-05:
I have tested the largest file (grey scale engraving) with [ioSender](https://github.com/terjeio/ioSender), aggressive buffering enabled.

150 steps/mm, max speed 25000mm/min, acceleration 1000 mm/sec2

iMXRT1062 (Teensy 4) controller, planner buffer set to 1000:
* Check mode streaming: 1:53, from SD card 0:11 (uses DMA).
* Run: 11:02, I've seen momentary feed rates above 17000mm/min.

LPC176x with BTT SKR 1.3 controller, planner buffer set to 150:
* Check mode streaming: 1:54, SD card not checked.
* Run:  I've seen a few momentary feed rates above 10000mm/min, I canceled the job at 40:00 due to the pauses (see below).

There was several pauses during the run, the job restarted by itself after a little while (a little less than 3 minutes). This is likely due to missed stepper interrupts causing the 32 bit timer used to not firing a new until the next wrap around. The timer runs at 24MHz so that matches well with 3 minutes. Most of these were at the end of the job.

After increasing the stepper interrupt priority, lowering the USB priority and fixing a bug in the SD card driver code I get these results:

* Check mode streaming: 1:58, from SD 2:45 (no DMA, single byte reads).
* Run: 13:42, I've seen a few momentary feed rates above 10000mm/min at the start of the job, many peaking at 18000 at the last 3rd of it. No pauses.
* SD card run: 13:52, I've seen a few momentary feed rates above 15000mm/min.

Not bad compared with the iMXRT1062 results.

Note that I have performed these tests without motors connected!

drom89, 2021-12-05:
> There was several pauses during the run, the job restarted by itself after a little while (a little less than 3 minutes).

I didnt wait for the resume, so i dont know if it will resume or not. Some times it stops and laser is still firing. 

> After increasing the stepper interrupt priority, lowering the USB priority and fixing a bug in the SD card driver code I get these results:

How can i do that ?

terjeio, 2021-12-05:
> How can i do that ?

Change priority in this line from 2 to 0.

https://github.com/grblHAL/LPC176x/blob/929596ccc370f56d15bfe5eba78fb26ca79c03e9/src/driver.c#L1306

Add this line before line 330:
`NVIC_SetPriority(USB_IRQn, 1);`

https://github.com/grblHAL/LPC176x/blob/929596ccc370f56d15bfe5eba78fb26ca79c03e9/src/usb_serial.c#L329-L330

What is your steps/mm setting for X? Number of microsteps per step? More is not necessarily better as it increases processor load.

drom89, 2021-12-05:
I have 80 steps per mm

readeral, 2022-01-19:
Reading through this thread, with the changes, are raster engravings now viable on the SKR 1.4T?

drom89, 2022-01-20:
To be honest i still didnt try it. I was busy with new version of mechanics.

drom89, 2022-02-14:
I have some troubles with TMC2130 overheating so im not sure if it is working as intended.

terjeio, 2022-02-16:
> I have some troubles with TMC2130 overheating

From the manual:
"The driver integrates a two level temperature sensor (120°C pre-warning and 150°C thermal shutdown) for diagnostics and for protection of the IC against excess heat."

When you write "overheating" do you mean thermal shutdown causing the motor(s) to stop moving?

I have now run a motor at 500mA for 1hr30 and the motor and the driver heatsink are both warm to the touch, I guess around 50°C. Will try a larger current setting later.

drom89, 2022-03-04:
So current situation when i try to engrave.
https://user-images.githubusercontent.com/5219063/156813970-b725f0a7-ecbc-4f25-b1e2-3ec48ffb91b9.mp4

drom89, 2022-03-04:
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:'$H'|'$X' to unlock]
error:9
G-code locked out during alarm or jog state.
[MSG:Caution: Unlocked]
ok
$$
$0=10.0
$1=250
$2=0
$3=1
$4=7
$5=7
$6=1
$7=0
$10=4095
$11=0.010
$12=0.002
$13=0
$14=7
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0
$21=0
$22=9
$23=3
$24=25.0
$25=1000.0
$26=250
$27=1.000
$28=0.000
$29=0.0
$30=1000.000
$31=0.000
$32=1
$33=50000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=1
$43=1
$44=3
$45=0
$46=0
$62=0
$63=3
$64=0
$65=0
$100=80.000
$101=80.000
$102=80.000
$110=66000.000
$111=30000.000
$112=10000.000
$120=6000.000
$121=4000.000
$122=1000.000
$130=600.000
$131=300.000
$132=200.000
$140=2000
$141=1000
$142=1000
$150=8
$151=16
$152=16
$200=0.0
$201=22.0
$202=50.0
$210=25
$211=50
$212=50
$220=0.0
$221=22.0
$222=50.0
$338=0
$339=0
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=100.0
$384=0
ok

[image]
[test_engrave.zip](https://github.com/grblHAL/LPC176x/files/8187815/test_engrave.zip)

drom89, 2022-03-04:
When i try graysacale image

[image]

drom89, 2022-03-04:
EDIT: videos above are with default BLOCK_BUFFER_SIZE

terjeio, 2022-03-04:
I'll take a look at this tomorrow.
Have you tried to lower the feed rate and/or lowering the acceleration?
X-axis acceleration looks a bit high to me - is the machine able to handle rapid (G0) motions without issues?

drom89, 2022-03-04:
Machine can take 1100mm/s on X Axis.

terjeio, 2022-03-06:
When I test with your settings for the x-axis my motor stalls. I have to reduce the acceleration substantially to make it work. At 500 mm/s2 the motor starts to sound ok and I can reach around 7000 mm/min feed rate with planner buffer size set to 150.
Note that this is with no load on the motor.

FYI running with your settings and counting step pulses with my CNC machine simulator no pulses are lost and the pulse train looks ok on my scope - albeit with a bit of jitter.

drom89, 2022-03-07:
I have around 1500mA on motors with 24V.
That was a problem with 2208 overheating :)

drom89, 2022-03-11:
I just tried it on second laser with older firmware and i dont get that stopping.

https://user-images.githubusercontent.com/5219063/157905491-56f3ad35-0737-49f2-b130-efe04eeb7ddc.mp4

terjeio, 2022-03-12:
$120 is 6000 for the new, the old 2000. Did you try reducing it for the new?

drom89, 2022-03-22:
So i tried cohesion3d smoothie cluster firmware and it seems i can get up to maximum speed 300mm/s grayscale engraving.
Link: https://forum.cohesion3d.com/t/improved-raster-speed-firmware-update-for-cohesion3d-boards/850

There is a link with changes to standard smoothie firmware [https://www.dropbox.com/s/bk44snjpu03dju1/SmooChanges.zip?dl=0](url) dont know if you can use any idea from it to GrblHAL.

terjeio, 2022-09-27:
An [experimental plugin](https://github.com/grblHAL/Plugins_laser/#lightburn-clusters) for cluster unpacking has been added to the laser plugins repo. An ongoing related discussion [here](https://github.com/grblHAL/core/discussions/195).

drom89, 2022-09-28:
Hello Terjeio, i already read about it, i catch the info in discussion of development of a new raster protocol. But thank you very much for letting me know.
Im testing ESP-WROOM-32 MCU for laser. Its quite cheap and it should be faster then LPC1769, we will see how it goes.
If it is possible to use it with GrblHAL i would be happy to test it.