grblHAL/LPC176x #15: Latest build will not compile (5 Errors)

Cncninja, 2021-11-27:
Thank you for your work, I was trying to update my grblhal to the latest branch and have the following errors when compiling. (usb, 5160,skr1.4turbo) any suggestions to fix this? Thank you

[image]

terjeio, 2021-11-27:
Looks like you have imported the project with a name different from the default. Change these include paths:

[image]

I'll fix that in the next commit.

Cncninja, 2021-12-12:
I must be doing something wrong i cant get this to ever compile
[image]

Cncninja, 2021-12-12:
Its looking for lusbd should it be libusbd ?

terjeio, 2021-12-12:
The linker is looking for `usbd_175x_6x_lib`, there is a `-l` in front of the library name.

There is at least one more path that has be changed:

[image]

If this does not help, close the project and open _.cproject_ in the root folder in an editor and replace `GRBL Driver LPC176x` with `${ProjName}`, then try again.

Cncninja, 2021-12-12:
I do not understand what to do with the -l

Cncninja, 2021-12-12:
${ProjName} seemed to work with no errors, just 23 warnings

Cncninja, 2021-12-12:
Should i worry about the top 4 warnings as they are new
[image]

terjeio, 2021-12-12:
> I do not understand what to do with the -l

Nothing, it is part of the linker command line.

> Should i worry about the top 4 warnings as they are new

If a working binary is produced you can ignore them. Or you may fix the _.cproject_ file as I described earlier. 

The other ones are due to sloppy code in the NXP libraries, I am not going to fix them.

closed 2022-01-21 by terjeio