grblHAL/LPC176x #16: Smoothieboard onboard stepper driver current settings

UniqueLemming, 2021-12-01:
Is there a way to set the current for the onboard stepper drivers on a Smoothieboard? I don't see anything in smoothieboard_map.h to set up the I2C based current controller. Am I missing something?

terjeio, 2021-12-01:
> Is there a way to set the current for the onboard stepper drivers on a Smoothieboard? 

No, not as of now. Perhaps the [MKS Sbase code](https://github.com/grblHAL/LPC176x/blob/master/src/mks_sbase.c) can be added or used as a starting point? This has configurable current control via I2C and MCP44XX pots.

Since I do not have a Smoothieboard to test with you will have to do that. I can provide guidance if needed.

UniqueLemming, 2021-12-03:
Thanks for the quick reply and offer for guidance. I see that the MKS Sbase machine file has an mks_sbase.c file that goes with it. It seems to have the code to enable the $14X settings that should allow setting of the current as well as the code to set up the I2C pots.

I've copied the mks_base.c file and named it my_machine.c. I've changed the #ifdef to BOARD_MY_MACHINE instead of BOARD_MKS_SBASE_13. 
I've also added 3 seemingly relevant lines from the mks_sbase_map.h to my_machine_map.h.

#define HAS_BOARD_INIT // comment out to disable driver current control
#define I2C_ENABLE 2
#define MCP44XX_I2C_ADDR        0b0101100

This did enable the $14X Current settings however the motors still have no power. I'm guessing that changes are needed in the .c file to account for the differences between these 2 boards. I'm not surprised it didn't work, but I'm not really sure what most of the code in that file is doing, or what might need to be changed. I was trying to look at the code from GRBL-LPC which is what I have been running on this machine, but my coding ability is pretty lacking and I'm having trouble making sense of that.

Any idea what I should try next?

terjeio, 2021-12-04:
> Any idea what I should try next?

I have made similar changes in the latest commit but forgot to tell you. Too much going on...

From the schematic the same I2C port and chip type is used as for the MKS SBase board. To debug this I would use a scope or a logic analyzer to look at the I2C signalling to check if it is ok. A simpler check is to measure the SDA and SDC voltages, if not 3.3V the I2C bus is stuck. Do you have acces to a scope or voltmeter?

UniqueLemming, 2021-12-04:
I don't have a scope, but I do have a voltmeter at least. I believe I have 3.3v on SDA, though I'm not 100% sure I checked the right pin. I'm not sure how to check SDC. I don't see a pin labeled SDC.

I did just recompile the firmware with your latest version and no changes except to uncomment the 2 necessary lines in the smoothieboard_map.h file. I have power on motor 4 oddly. Didn't even have motor 4 enabled in the firmware. I don't think there was any power on XYZ motors though. I could move them easily and changing the $140 settings didn't seem to have an effect.

terjeio, 2021-12-04:
> I don't see a pin labeled SDC.

Oops, it should be SCL. Brain not working early in the morning...

>  I have power on motor 4 oddly.

Ok, then the I2C address is wrong? - there are two pots on the board with IIRC A0 wired differently.
Try with `#define MCP44XX_I2C_ADDR 0b0101101`

The second pot is not supported in the code.

UniqueLemming, 2021-12-08:
I've been working on other things for a few days and didn't have time for this, but I've managed to get around to trying a few things. I believe that I do have 3.3v on both the SCL and SDA pins. Changing the address to 0b0101101 did not seem to change anything. I found the original address 0b0101100 in the config file of my working GRBL_LPC build, so I'm fairly confident that should be the one. 
The 4th motor thing is weird. My machine has 2 X-axis motors. The second motor is connected to the 4th axis motor driver. The GRBL_LPC build that I've been using didn't allow for ganged motors, so I physically jumped the 3 pins from the X-axis driver to the 4th motor driver.  When this set of jumpers is connected, I have no power to any of my motors with the GRBLHAL firmware. I only figured that out when I realized I had forgotten to disconnect the jumpers after switching firmware. When the jumpers are disconnected, The 4th motor has power, but no other motors do. The motor was also randomly moving around a very small amount. Probably due to floating pins? I had commented out the 4th motor in the map file.

I've been trying to compare the differences between the way GRBL_LPC and GRBLHAL set up the current, but it all looks like it should be working. Especially since this works of the mks_sbase boards. They look like they were setup the same way. There is a lot of this code that I don't really understand though.

Any thoughts on what I could try next?

terjeio, 2021-12-08:
> Any thoughts on what I could try next?

Try inverting the motor enable signals, $4=15. You may also measure the Vref voltage on the drivers coming from the pots to see if it can be changed by modifying the settings.

UniqueLemming, 2021-12-08:
$14 didn't make a difference, but I started putting in all my other $ settings and as soon as I got to $4=0 I heard the motors all engage. Guess I just needed to put in my settings this whole time..... face palm. Sorry for wasting your time. I really appreciate you taking the time to help me with this! 

It might be a good idea to add BOARD_MY_MACHINE to line 24 of mks_sbase.c now. Maybe something like:

#if defined(HAS_BOARD_INIT) && (defined(BOARD_MKS_SBASE_13) || defined(SMOOTHIEBOARD) || defined(BOARD_MY_MACHINE))

terjeio, 2021-12-08:
> It might be a good idea to add BOARD_MY_MACHINE to line 24 of mks_sbase.c now. Maybe something like:

I believe it is better idea to add `#define SMOOTHIEBOARD` to your _my_machine_map.h_ instead. This way _my_machine_map.h_ is not bound to any particular code and yours will "inherit" _mks_sbase.c_. Note that I have not tested this but I believe it will work.

UniqueLemming, 2021-12-08:
My only concern is that anyone else trying to use a Smoothieboard won't know to do that if they've created their own machine files. I'll give it a test soon though and at least verify that it works. I need to do some more work on this still anyway. My machine still isn't moving properly and I really want to get the X axis to be ganged properly on this machine, so I'll be doing a recompile with that next. Now that I know I can get the onboard stepper drivers working, I can move on to the rest of the setup.

terjeio, 2021-12-08:
> My only concern is that anyone else trying to use a Smoothieboard won't know to do that if they've created their own machine files. 

_my_machine_map.h_ is for private maps and this file will never be part of the distribution. It is not possible to activate board specific extensions for that without creating problems for those who do not need them. Maybe a comment in the Smootieboard map providing some guidance would help?

closed 2021-12-13 by UniqueLemming

terjeio, 2021-12-13:
FYI I have added support for the 5th axis in the latest commit, please report back if this breaks anything.

UniqueLemming, 2022-05-24:
Okay, I finally got some time to finish upgrading my laser cutter's smoothieboard to run on grblHAL. I figured I would just add a follow up to this incase anyone wanted to know. Current control of the onboard stepper drivers is working nicely. With a few modification to the map file I was able to get my X axis ganged and auto squaring nicely. 4th motor is confirmed working. My board doesn't have the 5th axis, so I can't check it, but it should work fine. You do need to flip the motor enable pins with $14 though.