grblHAL/LPC176x #18: Spindle enable and direction not changing state

Cncninja, 2021-12-18:
Hello, the SKR 1.4 turbo has pins 1.23 and 1.24 defined as spindle enable and direction. I'm using I/O sender edge i can select turn on the spindle CW or CCW and pins 1.23 and 1.24 do not change state. Between the pins and +5V I get 5V and between pins and ground, i get 0V So the pin is Low/ground all the time.  Flood and mist seam to work in the same EXP1 connector

[image]

[image]

terjeio, 2021-12-19:
These pins works with a SKR 1.3 (same mapping), is `$31=1` (laser mode enabled)? If so the spindle output is only switched on when there is motion. Nothe that when CCW is on, jogging or G0 moves will _not_ switch the spindle on in laser mode.

Cncninja, 2021-12-22:
confirmed it was in normal mode, not Laser, but based on your input I loaded a file and hit cyc start and the spindle did change state. i stopped the program and could still turn on and off the spindle. if I do a reset tho it does not work till the code is run then stopped.

terjeio, 2021-12-22:
Are you using the correct pin map?
Check with `$i` and `$pins` commands.

closed 2023-01-29 by terjeio