grblHAL/LPC176x #19: grbl file error on compile

Cncninja, 2022-01-01:
[image]

terjeio, 2022-01-01:
grbl folder is empty? [submodules missing?](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL#using-git-command-line)

[image]

Cncninja, 2022-01-01:
Yes, that was the issue, forgot to click clone submodules,  but the spindle does not work with the new build pin 1.23 if I pop in the old bin works fine.  no selection changes between builds

terjeio, 2022-01-02:
Which board map? From the error list above it looks like you are compiling for SKR E3 Turbo which has spindle enable on 0.16.

Cncninja, 2022-01-02:
skr1.4turbo, it's an odd behavior it will work for a bit but then just stop responding, the flood and mist still work and if a do reset the spindle works again for a few acuations, I can see in the console the M3 s1000 is being sent but the SKR just ignores it. if I do a pin invert it does change state so the board is under control of the output.

terjeio, 2022-01-05:
I am not able to replicate the spindle behaviour. Testing with this version and board:
```
[DRIVER VERSION:211222]
[BOARD:BTT SKR V1.3]
```

Can you list the exact steps that triggers it?
Is it the PWM or the spindle on output that fails?

Your settings may be helpful as well.

closed 2023-08-26 by terjeio