grblHAL/LPC176x #20: S32 alternatives ?

Frogowitsch, 2022-01-17:
Hi,
I'm a MAC user and there is no S32 Studio available for my OS. Is there any possibility to compile this outside S32 ? I need to compile the sources for my MKS Base 1.3 and cant find any possibility on a Mac. Installing Windows in a VM is no Option, since i would need to buy Windows first and being a student, thats no option. 
What can i do ? If there is no possibility, could someone please compile it and upload the image file ? Hope you can help me.

Frogowitsch, 2022-01-17:
OK, i see there is a version of MCUXpresso available. Will try this. Seems to be the way, sorry to bother

Frogowitsch, 2022-01-17:
No way either, LPC1768 is not supported (MCUXpresso cant find any SDK for this Cortex M3. COuld someone help please ? I hope there is a way to use the code, would hate to give up and buy another board...

terjeio, 2022-01-17:
> LPC1768 is not supported

I am using MCUXpresso 11.3.1 for development - and the SDK files needed are included in the download. So you should be able to compile after importing.

Frogowitsch, 2022-01-17:
Ok, i see. Thank you very much. I was searching for the SDK files before trying to compile :). LOL. Compiling fine, thank you for help.

Frogowitsch, 2022-01-18:
Where can I ask questions about this driver ? I got all nearly working, my only problems are, that I dont know for example how to configure the stepper current now. The SBASE 1.3 has no possibility to do this with trimmers and i cant find any function inside GRBL to set the current... Is there a forum you can recommend, i dont want to spam this thread ;).

terjeio, 2022-01-18:
> Where can I ask questions about this driver ? 

Here.

>  how to configure the stepper current

By altering $-settings.

> Smoothie had an option in its configs.

Same with grblHAL, and there is help built in. `$help` sent to the controller that I happen to have connected outputs this:
```
Help topics:
 Commands
 Settings
 Aux ports
 Axis
 Control signals
 Coolant
 General
 Homing
 Jogging
 Limits
 Probing
 Spindle
 Stepper
 Tool change
 X-axis
 Y-axis
 Z-axis
ok
```
`$help x-axis` this:
```
---- X-axis:

$100: X-axis travel resolution in step/mm

Travel resolution in steps per millimeter.

$110: X-axis maximum rate in mm/min

Maximum rate. Used as G0 rapid rate.

$120: X-axis acceleration in mm/sec^2

Acceleration. Used for motion planning to not exceed motor torque and lose steps.

$130: X-axis maximum travel in mm

Maximum axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.
ok
```

You should  find the setting number for x-axis current in the output you get since it is generated from the active configuration.

Frogowitsch, 2022-01-18:
Thank you terjeio, its running now. Perfect.  Very nice, you helped so much. Thanks

closed 2022-01-18 by Frogowitsch