grblHAL/LPC176x #21: Second serial interface

Frogowitsch, 2022-01-18:
Hi, GRBLHAL runs very nice on the MKS BASE 1.3, thanks for this driver and all the help to get it starting.
I've got a last question. I would like to create an external jogger and need a second serial port for this. There is an expansion port on this board, that is used for the external TFT. It uses serial communication, so there must be a pair of RX/TX signals, but my attempts all failed until now. Is there a possibility to access the second serial port with GRBLHAL ? This would make it much easier to communicate with the board.

Frogowitsch, 2022-01-19:
Or ist a second port not required ? Can i replicate the first serial port to two more pins. So i have the serial console on usb AND on two other pins (rx/tx) simultaneously. Is this possible with this driver ?

terjeio, 2022-01-19:
Are you using USB as the main communications channel?

> Can i replicate the first serial port to two more pins. 

Yes, with plugin code a lot of fancy stuff can be done. FYI _MPG mode_ for the iMXRT1062 driver replicates the USB output stream to the UART and accepts input from it - either routed to the keypad plugin or by taking over full control.

If _MPG mode_ is what you need it can easily be added to this driver since I have moved most of the stream handling code to the core. 

Additional UART ports (up to max 4 for now) can be added by user code and made available for plugins via a discovery/claim API.
What I have not done is to update all the drivers to take full advantage of the functionality offered by core - I would like to have help with that.

Frogowitsch, 2022-01-19:
Hi.
To be honest I‘m not sure what exactly would be best, i can write what my intension is.  
I‘m creating a handwheel that i would like to interact with the board and not the sender. I use the ESP32 atm and output and input all over rx/tx. This way i can now move the motors (joggling with g0 absolute coordinates, so the mcu in the handwheel counts the pulses of the wheelencoder). And i can also start/stop. I‘m about to implement more functions, all easy serial input/output with micropython. 
But atm i can only use it as a replacement for the gcode sender software on my PC. 
it would be very nice to use both. Gcode sender on the PC and when needed i take the wheel and jog around and zero an axis or start the program. 
So i think MPG mode, full control is the way to go. Do you also think ?
Where can I configure this ? Do i need to write code for this or just config ? I would love to help, but I‘m only a hobby coder and my knowledge does not go over simple mcu programs.

Frogowitsch, 2022-01-19:
Yes, USB is my channel connected to the PC with Gcode sender. 
Atm my Handwheel only functions if I disconnect USB from the PC and connect the handwheel with an serial->USB converter.

terjeio, 2022-01-23:
FYI I have added the MPG option to this driver in the latest commit, but not tested it yet. You can test?

Frogowitsch, 2022-01-23:
Yes, i will test it today. Thank you very much.

Frogowitsch, 2022-01-23:
To be honest, I failed to configure it. The Aux-1 port uses P0.2 and P0.3 pins for in and output, but in the mks_base config, I only found one port and one pin for the MPG_MODE. Or can i configure it somewhere else ? Did you implement the full control mode, so whole g codes can be send and read over rx/tx ? Thank you for the work, youre the best.

Frogowitsch, 2022-01-23:
Ok, the MPG Mode has nothing to do with normal serial port communication ? I found the MPG mode commands, but still do not understand how the whole communication functions. So i will have to rewrite my whole mcu code for the handwheel, so it sends the mpg commands. Will do this tomorrow and test it then... 
Is there any kind of tutorial or further documentation available for the mpg mode besides reading the sourcecode ? Its hard for a non-coder.

terjeio, 2022-01-23:
The MPG mode pin is for switching between input sources. When in MPG mode an UART port is given full control instead of the USB port. The `$pins` command should show which pins assigned to RX/TX. An explanation for [MPG mode](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack#how-it-works) can be found here. [This discussion](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack/discussions/6) may also be of help.

FYI later versions allows the use of the second port for input to the [keypad plugin](https://github.com/grblHAL/Plugin_keypad) - this uses single character commands for many functions but does not support gcode input.

Frogowitsch, 2022-01-24:
Thank you, i will test it tomorrow.

Frogowitsch, 2022-01-24:
This is what the $pins command show me:
[PIN:P1.23,Probe]
[PIN:P0.23,Reset]
[PIN:P0.24,Feed hold]
[PIN:P0.25,Cycle start]
[PIN:P1.24,X limit min]
[PIN:P1.22,MPG mode select]
[PIN:P1.26,Y limit min]
[PIN:P1.28,Z limit min]
[PIN:P2.0,X step]
[PIN:P2.1,Y step]
[PIN:P2.2,Z step]
[PIN:P2.3,X2 step]
[PIN:P0.5,X dir]
[PIN:P0.11,Y dir]
[PIN:P0.20,Z dir]
[PIN:P0.22,X2 dir]
[PIN:P0.4,X enable]
[PIN:P0.10,Y enable]
[PIN:P0.19,Z enable]
[PIN:P0.21,X enable]
[PIN:P2.11,Spindle on]
[PIN:P4.28,Spindle direction]
[PIN:P2.6,Flood]
[PIN:P2.7,Mist]
[PIN:P0.6,SD card CS]
[PIN:P0.1,SCK]
[PIN:P0.0,SDA]
ok

No mention of RX/TX
If i use ? I get this:

Run|MPos:365.335,0.000,40.000|Bf:34,1024|FS:5000,0|Pn:PYZHS>
ok
<Run|MPos:450.350,0.000,40.000|Bf:34,1024|FS:5000,0|Pn:PYZHS>
ok
<Idle|MPos:500.001,0.000,40.000|Bf:35,1024|FS:0,0|Pn:PYZHS>
ok

There is no new |MPG: section. If i pull pin 1.22 (MPG Mode select) high, nothing changes. Low also nothing.
Of course its possible, that i understand the concept of mpg wrong, but after 3 hours i give up for today ;). No idea.

Frogowitsch, 2022-01-24:
In MPG normal mode I'm supposed to listen to the stream on my UI MCU. This would be nice, so i can show DRO on its OLED. But where can i configure the pins, where i can listen to this. Or do i need to make a parallel connection to the USB RX/TX ? Of course this would lower the voltage on this lines... And i get a "port already busy" error ofc.
As a non-coder its hard to understand the code.

terjeio, 2022-01-26:
I have now cleared my desk and found time to test this - and there were several bugs to fix. I have just committed the changes.
The mode select pin has to be on an interrupt capable port, only 0 and 2 are, so I have defined that to [P2.0 on the Servos header](https://github.com/grblHAL/LPC176x/blob/d1280f4a98290f62089bf83d56258e35e1fbeb93/src/btt_skr_1.3_map.h#L174-L178) for SKR 1.3 that I tested with. For the SKS board P0.23 - P0.26 on the TH header could be used (there is a filter cap on these inputs so perhaps no the best) or better one of the P0 pins on the EXP1 or EXP2 headers. You have to modify the board map file for the one you select.

There is one issue to be aware of when the primary port is USB, if data is replicated to that when in MPG mode (it is is per default) the controller may hang if there is no client connected. What is needed is a reliable way to find out if a client is connected and not write to the USB port if not so - I have yet to figure out how to do that and if it is even possible. BTW it is not good enough to detect that the USB cable is connected which is simpler...

Output from `$pins` when MPG_ENABLED is uncommented in [my_machine.h](https://github.com/grblHAL/LPC176x/blob/d1280f4a98290f62089bf83d56258e35e1fbeb93/src/my_machine.h#L40) for SKR 1.3:
```
[PIN:P0.12,Probe]
[PIN:P0.18,Reset]
[PIN:P0.16,Feed hold]
[PIN:P0.15,Cycle start]
[PIN:P1.29,X limit min]
[PIN:P2.0,MPG mode select]
[PIN:P1.27,Y limit min]
[PIN:P1.25,Z limit min]
[PIN:P2.2,X step]
[PIN:P0.19,Y step]
[PIN:P0.22,Z step]
[PIN:P2.6,X dir]
[PIN:P0.20,Y dir]
[PIN:P2.11,Z dir]
[PIN:P2.1,X enable]
[PIN:P2.8,Y enable]
[PIN:P0.21,Z enable]
[PIN:P1.23,Spindle on]
[PIN:P1.21,Spindle direction]
[PIN:P3.25,Flood]
[PIN:P3.26,Mist]
[PIN:P0.3,RX]
[PIN:P0.2,TX]
ok

```

Frogowitsch, 2022-01-27:
Thanks a lot. Its running now. I also got my RX/TX pins.
One last thing: Where can I set the Port/Pins for RX/TX ? Cant find it in the files... I was thinking a bit about it and now it seems best, to have RX/TX and MPG mode pin all on the J7 connector. It has 5V, GND, P0.15, P0.16, P0.17. Like a dream for this purpose. This way its possible to have RX/TX for handwheel AND the mode select all on one connector. And its Port 0, so MPG selector can be on interrupt. Ofc i found the MPG port/pin, but i cant find the uart one. Plz help me with this last problem.

terjeio, 2022-01-27:
> Where can I set the Port/Pins for RX/TX ?

The port is defined [here](https://github.com/grblHAL/LPC176x/blob/d1280f4a98290f62089bf83d56258e35e1fbeb93/src/serial.h#L30-L32) and the pins [here](https://github.com/grblHAL/LPC176x/blob/d1280f4a98290f62089bf83d56258e35e1fbeb93/src/serial.c#L40-L44).

> This way its possible to have RX/TX for handwheel AND the mode select all on one connector.

From the user manual, page 118:

|PINSEL0|Pin name|Function when 00|Function when 01|Function when 10|Function when 11|Reset value|
|-------|--------|----------------|----------------|----------------|----------------|-----------|
|31:30|P0.15| GPIO Port 0.15|TXD1|SCK0|SCK|00|

|PINSEL1|Pin name|Function when 00|Function when 01|Function when 10|Function when 11|Reset value|
|-------|--------|----------------|----------------|----------------|----------------|-----------|
|1:0|P0.16|GPIO Port 0.16|RXD1|SSEL0|SSEL|00|
|3:2|P0.17|GPIO Port 0.17|CTS1|MISO0|MISO|00|

Frogowitsch, 2022-01-27:
Thanks, now i know where to change the pins. I dont understand the Pinsel0 and 1 from user manual, do you think I. An use 0.15, 0.16 and 0.17 for TX, RX and MPG Pin ? 

I‘ve got my DRO running fine, all perfect and will implement Gcodes and joggling later. I update DRO about 10 times per second.  Do you think thats too often ? Looks smooth on a OLED screen, but is there any downside if I read out the values so often per second ? (My feeling says no, but still better ask than degrade performance).

terjeio, 2022-01-27:
> I dont understand the Pinsel0 and 1 from user manual, do you think I can use 0.15, 0.16 and 0.17 for TX, RX and MPG Pin ?

Yes, try with changing the pin config to this:

```
static const PINMUX_GRP_T uart_pinmux[] = {
    {0, 15, IOCON_MODE_INACT | IOCON_FUNC1}, /* TXD0 */
    {0, 16, IOCON_MODE_INACT | IOCON_FUNC1}  /* RXD0 */
};
```

and the [port definitions](https://github.com/grblHAL/LPC176x/blob/d1280f4a98290f62089bf83d56258e35e1fbeb93/src/serial.h#L30-L32) to UART1.

>  I update DRO about 10 times per second. Do you think thats too often ?

I do not think so, BTW the default update rate most senders use is 5 times per second.

> I‘ve been using LinuxCNC for a long time, because of the slow 8 bit GRBL mcus, but now there is no more reason to use a PC as step controller. 

Interesting to hear that - grblHAL is getting closer in capabilities but can never (?) reach the same level as LinuxCNC due to limited RAM in microcontrollers.

Frogowitsch, 2022-01-27:
If you take LinuxCNC with a Mesa Card, then yes, its still smoother and faster than GRBLHAL, but also A LOT more expensive. LinuxCNC with a parallel bob is not really superior any more (at least it seels like that). The PC struggles, if not helped by an FPGA. 
Perhaps an FPGA implementation of GRBLHAL with some MBs of RAM would be the way to go if you want to get it to LinuxCNC levels.

Frogowitsch, 2022-01-28:
Ok, runs fine now. Thank you once again. Good work !

closed 2022-01-28 by Frogowitsch