grblHAL/LPC176x #22: Cant change the microsteping value

drom89, 2022-01-28:
Hi, i cant change microstepping value from 16.
Every time i change it, it get back to 16.
Board BTT SKR 1.4T
I tried it with TMC2209 and TMC2130

What im doing wrong ?

terjeio, 2022-01-29:
Please update to the latest version and try again, if still a problem then report back.

drom89, 2022-02-03:
After updating, it still go back to 16.

[image]

terjeio, 2022-02-04:
Settings does not "take" unless the drivers are online, and I see now that they are not initialized on startup as they should. I will look into this a bit later.
Send a `M122I` command after startup to force initalization, that helps for me.
The drivers has to be enabled with `$338`, and `M122` should return driver status if they are online.

drom89, 2022-02-04:
So it looks like my drivers BTT TMC 2130 are not communicating.
And i need to desolder this resistor to switch it to SPI mode.

[image]

drom89, 2022-02-04:
So i desolder the SPI and connected CFG4 and CFG5 see the picture below, but the connection is still not working :(
[image]

terjeio, 2022-02-04:
Set `$338` to `1` to enable only the X driver. Then send `M122I` followed by `M122`. Do you still get the error message about not beeing able to communicate with the driver?

I am testing with a SKR 1.3 board and this fails to communicate with the driver on a "normal" startup, but when restarted by the debugger it works (withouh issuing `M122I`). I am trying to figure out why...

drom89, 2022-02-04:
[image]

drom89, 2022-02-04:
I tried 2nd driver and it seems to work
[image]

terjeio, 2022-02-04:
Ok, so you have a bad driver?
With the second responding you should now be able to set microstepping for it.

drom89, 2022-02-04:
Yes it was faulty driver.
Its working now. 
Thank you

drom89, 2022-02-04:
BTW motors seems to be slower (lower RPM) when i compare it to STEP/DIR mode.
Do you have any advice on settings how to get maximum RPM ?

terjeio, 2022-02-04:
Decrease microstepping settings and/or increase steps/mm settings.

terjeio, 2022-02-04:
Add this line [here](https://github.com/grblHAL/LPC176x/blob/af2de58485d5435c9ea3ab05d2253b70a3d1c942/src/btt_skr_1.x.c#L180) to get the drivers to initialize on startup:
`hal.delay_ms(100, NULL);`

The delay allows the drivers to "boot up" before attempting to talk to them.

terjeio, 2022-02-04:
> Do you have any advice on settings how to get maximum RPM ?

The settings normally used for max RPM is maximum (feed)rate, $110, $111 and $121. Microstepping and steps/mm is for calibrating real movement to commanded movement.

closed 2022-02-08 by drom89