grblHAL/LPC176x #25: SKR 1.4 turbo Serial RX/TX Wifi

VeXed2, 2022-04-08:
Hello, I am trying to use [this](https://www.biqu.equipment/products/esp-01s-wifi-module-esp8266-for-skr-pro?_pos=1&_sid=f301448de&_ss=r) to use the board with Wifi instead of USB. I've commented "#define USB_SERIAL_CDC" in my_machine.h, I've put these: 
-     {4, 28, IOCON_MODE_INACT | IOCON_FUNC1}, /* TXD3 */
-     {4, 29, IOCON_MODE_INACT | IOCON_FUNC1}  /* RXD3 */
-     #define SERIAL_IRQHandler UART3_IRQHandler
-     #define SERIAL_MODULE     LPC_UART3
-     #define SERIAL_MODULE_INT UART3_IRQn
 but I can't manage to connect through serial, the web page for esp works, I've set it to GRBL but can't manage to get data from it. Anyone knows how I can make this work?

terjeio, 2022-04-10:
Try with `IOCON_FUNC3` instead of `IOCON_FUNC1` for the pin mapping. Ref table 86 in the user manual (UM10360).

VeXed2, 2022-04-10:
That fixed it. Thank you very much :)

closed 2022-04-10 by VeXed2