grblHAL/LPC176x #26: Autosquaring with Re-Arm and Ramps 1.6

regentonne5, 2022-04-15:
Hello, I am trying to convert my cnc maschine (mpcnc) to grblHal but I am struggeling with ganged axis / auto squaring.
I want to use dual endstops for x and y axis and so my question is how to define that grbl uses the limit_pin_max as a second limit_pin. 
Is it enough to just enable #define x_ganged and #define x_auto_square in the my_maschine.h file or do i have to change something else, because on the frontpage it says that the board map file has to be edited. 

Thanks in advance!

terjeio, 2022-04-15:
If you want auto squaring then change this code:

https://github.com/grblHAL/LPC176x/blob/d05d1db721d0b780b3d327f036ddb36518ac9623/src/ramps_1.6_map.h#L104-L130

to:

```
// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_STEP_PN              2
#define M3_STEP_PORT            port(M3_STEP_PN)
#define M3_STEP_PIN             0   // Due Digital Pin 26
#define M3_DIRECTION_PN         0
#define M3_DIRECTION_PORT       port(M3_DIRECTION_PN)
#define M3_DIRECTION_PIN        5   // Due Digital Pin 28
#define M3_ENABLE_PN            0
#define M3_ENABLE_PORT          port(M3_ENABLE_PN)
#define M3_ENABLE_PIN           4   // Due Digital Pin 24
#if !LIMIT_MAX_ENABLE
#define M3_LIMIT_PN             1
#define M3_LIMIT_PORT           port(M3_LIMIT_PN)
#define M3_LIMIT_PIN            25  // Due Digital Pin 2
#endif
#endif

// Define ganged axis or B axis step pulse and step direction output pins.
#if N_ABC_MOTORS == 2
#define M4_AVAILABLE
#define M4_STEP_PN              2
#define M4_STEP_PORT            port(M4_STEP_PN)
#define M4_STEP_PIN             8   // Due Digital Pin 36
#define M4_DIRECTION_PN         2
#define M4_DIRECTION_PORT       port(M4_DIRECTION_PN)
#define M4_DIRECTION_PIN        13  // Due Digital Pin 34
#define M4_ENABLE_PN            4
#define M4_ENABLE_PORT          port(M4_ENABLE_PN)
#define M4_ENABLE_PIN           29  // Due Digital Pin 30
#if !LIMIT_MAX_ENABLE
#define M4_LIMIT_PN             1
#define M4_LIMIT_PORT           port(M4_LIMIT_PN)
#define M4_LIMIT_PIN            27  // Due Digital Pin 15
#endif
#endif
```

and uncomment 
https://github.com/grblHAL/LPC176x/blob/d05d1db721d0b780b3d327f036ddb36518ac9623/src/my_machine.h#L51

If you only want to enable max limit switches only then uncomment
https://github.com/grblHAL/LPC176x/blob/d05d1db721d0b780b3d327f036ddb36518ac9623/src/my_machine.h#L44

I have not made it simple to mix auto squared axes with axes having a max limit switch. You will have to modify the board map file for that.

regentonne5, 2022-04-17:
It works!
Thank you for your fast reply. I would like to make a small donation to you for helping me and for your amazing work with grblHAL. What would be the best way to donate?

closed 2022-04-20 by regentonne5