grblHAL/LPC176x #27: where is the upload.ino

killrob, 2022-04-16:
For the ones like me that wants to use Arduino ide, where is the upload.ino file? i all repos that i downloaded there is no "example" folder containing this file....

terjeio, 2022-04-17:
Compiler/IDE used for building is shown as the last item in the [driver overview lists](https://github.com/grblHAL/drivers). Some can also be built using PlatformIO.
I have no plans to add Arduino IDE support for this driver.

killrob, 2022-04-17:
> Compiler/IDE used for building is shown as the last item in the [driver overview lists](https://github.com/grblHAL/drivers). Some can also be built using PlatformIO. I have no plans to add Arduino IDE support for this driver.

ok, i want to use it on a SKR V1.4 that is similar to the TURBO is a 1768 but...

in [this](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL) page you talk about comiling with Arduino IDE or ECLIPSE, no word about using VScode with platformio. I'm quite lost and i don't understand.

terjeio, 2022-04-18:
> i want to use it on a SKR V1.4 that is similar to the TURBO is a 1768 but...

Then you can find a [pre-built binary here](http://www.io-engineering.com/downloads/).

>  no word about using VScode with platformio

I do not use platformio myself, other users does and has added platformio.ini via pull requests for some of the drivers. The build environments I support are those listed in the driver overview.

SwarleyAUS, 2024-10-21:
Sorry to resurrect this thread, but there does not seem to be any pre-built binaries for the SKR 1.4 in the listed location anymore. Can you advise how to build the firmware? I can see no valid arduino library being present despite the compiling guide saying otherwise.

terjeio, 2024-10-21:
@SwarleyAUS The [driver overview](https://github.com/grblHAL/drivers) specifies MCUExpresso which is Eclipse based. So there is no .ino for this driver.

SwarleyAUS, 2024-10-21:
ah ok thank you, can you recommend any guides... not familiar with that IDE and how to build a hex

terjeio, 2024-10-22:
FYI I do not get notified about edits to comments so better to create a new one if there is any significant changes.

> Where do UART pins get assigned?

[Here](https://github.com/grblHAL/LPC176x/blob/75a181175e929a494128d2a64d1d21796c5a95f6/src/serial.c#L43-L46) and [here](https://github.com/grblHAL/LPC176x/blob/75a181175e929a494128d2a64d1d21796c5a95f6/src/serial.c#L55-L58).

> What does #define SERIAL_MOD 3 do in my board map?

Switches to UART 3 for the serial port.

> How do I define TRINAMIC_UART_ENABLE to be true so I can use UART with the drivers?

You cannot, only soft UART mode is supported, not hardware UART. Soft UART mode is enabled by enabling the 2209 driver in _my_machine.h_ by uncommenting [this line](https://github.com/grblHAL/LPC176x/blob/75a181175e929a494128d2a64d1d21796c5a95f6/src/my_machine.h#L60).

> Is there a manual covering all of this?

No, basically only the comments in _my_machine.h_. However, plugins may have more details available in the readme file.

SwarleyAUS, 2024-10-23:
Great, thankyou, that helps. However UART is still not working. pins 28 and 29 are the USB UART pins I believe, so this is what I have:

in my_machine.h:
#ifndef USB_SERIAL_CDC
#define USB_SERIAL_CDC           0 // Comment out to use UART communication.
#endif

which should set SERIAL_MOD to 3 per this code in the SKR 1.4 turbo map:
#if USB_SERIAL_CDC == 0
#define SERIAL_MOD 3    // WiFi port
#endif

and that should set the active serial port to pins 28 and 29 per this code in serial.c:

#if SERIAL_MOD == 3
...
    {4, 28, IOCON_MODE_INACT | IOCON_FUNC3}, /* TXD3 */
    {4, 29, IOCON_MODE_INACT | IOCON_FUNC3}  /* RXD3 */
};

correct? so do you have any idea why a serial port is not being presented? its being converted to a .cur file by the board, and the uart port works because ive tested BTT's 'test' firmware - with that a serial port presents in device manager straight away.

terjeio, 2024-10-24:
> However UART is still not working. pins 28 and 29 [see edit] are the USB UART pins I believe

There is no such thing as USB UART for this driver. When USB_SERIAL_CDC  is enabled the MCU itself provides the USB stack, not an external chip connected via an UART channel.
If you disable USB_SERIAL_CDC then you have to connect an UART <> USB converter, eg a FT232 based one, to the UART pins if you want to talk with the card via a USB connection (virtual serial port).

SwarleyAUS, 2024-10-24:
ok so USB is what I want, I want it to present as a COM port in windows. Right now it doesn't and idk why. What I'm noticing in the IDE is that it thinks that SERIAL_MOD == 3 isn't true, because that block of code is greyed out(?), why would that be? 

`#define BOARD_BTT_SKR_14_TURBO` is uncommented and 
`#if USB_SERIAL_CDC == 0 
#define SERIAL_MOD 3` is not greyed out in the board map.

[image]

SERIAL_MOD 3 is set in the board map. Is this an issue or a red herring? Any ideas how I can get *USB* working?

terjeio, 2024-10-24:
> Any ideas how I can get USB working?

#define USB_SERIAL_CDC 1` in [my_machine_map.h](https://github.com/grblHAL/LPC176x/blob/ac5e4a41b374a0cd27cb75c8dd03daf9025f9b34/src/my_machine.h#L36).

SwarleyAUS, 2024-10-24:
sorry in my_machine.h or in the board map header file? I've tried both and neither seem to rectify the issue though.

Seeing as we've been talking at cross purposes, where can I check the *USB* pins are mapped correctly?

terjeio, 2024-10-24:
> sorry in my_machine.h or in the board map header file?

in my_machine.h

>  where can I check the USB pins are mapped correctly?

I would guess somewhere in the NXP framework, and I believe there is only one set of pins that can be used. So you will have to dig into that and also consult the datasheet to see if more than one set can be used. However, when I have tested this board USB comms has been working for me without worrying about the pin mappings - so I assume they are ok.

terjeio, 2024-10-24:
> So I shouldn't need to change SERIAL_MOD from 3 to anything else?

No, it has nothing to do with serial over USB. The UART port may be used for other purposes such as a MPG.

If you flash with the SD card you have to build a [bootloader version](https://github.com/grblHAL/LPC176x/blob/master/README.md). You did that?

SwarleyAUS, 2024-10-24:
amazing! that's done it. I didn't realise putting the binary on the card was considered "Smoothieboard DFU-bootloading". thank you so much for your help.