grblHAL/LPC176x #28: Help needed with compiling please

nutbox76, 2022-04-16:
Hi there,
I was wondering if someone could help me,
I have a k40 which I’ve installed a skr v1.4 turbo board into and have it running with smoothieware but I am looking to install grblhal because it looks like a better firmware .
The problem I am having is that I am falling at the first hurdle and can’t for the life of me figure out how to compile it via platformio.
I have opened a new project with the relevant mcu selected which is LPC1769 and I have downloaded the grblhal core and the driver lpc176x files and then from there I don’t really know where I should be going.
So if there is any could write ups that are easy to follow or YouTube videos that could help please point me to them,or maybe someone has the firmware already compiled and don’t mind me trying it that would be great .
I have setup marlin before on 3d printers and smoothie as I said .
Thank you Dazz

terjeio, 2022-04-17:
Check this [issue](https://github.com/grblHAL/LPC176x/issues/12).
You can find a prebuilt binary [here](http://www.io-engineering.com/downloads).

nutbox76, 2022-04-17:
Thank you I shall check it out later on tonight

nutbox76, 2022-04-22:
Hi there
I have flashed the linked binary but as soon as i switch the machine on X and Y steppers lock up and the machine wont home and the terminal comes up with this.
ALARM:8
Homing fail. Cycle failed to clear limit switch when pulling off. Try increasing pull-off setting or check wiring.
ALARM:8
Homing fail. Cycle failed to clear limit switch when pulling off. Try increasing pull-off setting or check wiring.
ok
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:'$H'|'$X' to unlock]
[MSG:Caution: Unlocked]
These are my settings so far and i know i am in over my head really but how can i sort this out please.
$0=10.0
$1=25
$2=0
$3=3
$4=0
$5=0
$6=0
$7=0
$10=3
$11=0.010
$12=0.002
$13=0
$14=3
$15=0
$16=0
$17=0
$18=0
$19=0
$20=1
$21=0
$22=3
$23=3
$24=200.0
$25=800.0
$26=250
$27=4.000
$28=0.100
$29=0.0
$30=1000.000
$31=0.000
$32=1
$33=5000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=0
$43=1
$44=4
$45=3
$46=0
$62=0
$63=3
$64=0
$65=0
$80=1.000
$81=0.010
$82=0.000
$84=0.000
$85=10.000
$90=0.000
$91=0.000
$92=0.000
$95=0.000
$100=157.575
$101=157.575
$102=250.000
$110=12000.000
$111=12000.000
$112=500.000
$120=4000.000
$121=4000.000
$122=10.000
$130=210.000
$131=300.000
$132=200.000
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=100.0
$347=5.0
$348=2.500
$349=25.000

I am using a skr v1.4 turbo board in a k40 if thats any help.
Thank you

nutbox76, 2022-04-22:
scrub that i have now managed to get it to home all be it a bit clunky at the moment ,i installed iosender which makes it easier to adjust the setting.

dannyb131, 2024-02-29:
dose anyone have a copy of the  prebuilt binary for the SKR 1.4

terjeio, 2025-04-16:
The file has been removed since it was out of date.

closed 2025-06-02 by terjeio