grblHAL/LPC176x #30: LPC176x build fails with TMC2209 drivers

sn0wrider, 2022-10-24:
Hello everyone, I'm trying to make a firmware for skr 1.3 with TMC2209 drivers, but I'm have error
compiler MCUXpresso IDE v11.6.1_8255. does anyone know how to solve the problem?

../src/tmc_uart.c:375:34: warning: passing argument 2 of 'hal.enumerate_pins' from incompatible pointer type [-Wincompatible-pointer-types]
  375 |         hal.enumerate_pins(true, add_uart_pin);
      |                                  ^~~~~~~~~~~~
      |                                  |
      |                                  void (*)(xbar_t *)
../src/tmc_uart.c:375:34: note: expected 'pin_info_ptr' {aka 'void (*)(xbar_t *, void *)'} but argument is of type 'void (*)(xbar_t *)'
../src/tmc_uart.c:375:9: error: too few arguments to function 'hal.enumerate_pins'
  375 |         hal.enumerate_pins(true, add_uart_pin);
      |         ^~~
At top level:
../src/tmc_uart.c:83:24: warning: 'outputpin' defined but not used [-Wunused-variable]
   83 | static output_signal_t outputpin[] = {
      |                        ^~~~~~~~~
make[1]: *** [src/subdir.mk:66: src/tmc_uart.o] Error 1
make[1]: *** Waiting for unfinished jobs....
Finished building: ../src/serial.c
 
Finished building: ../src/usb_serial.c
 
Finished building: ../src/driver.c
 
make: *** [makefile:48: all] Error 2
"make -r -j12 all" terminated with exit code 2. Build might be incomplete.

15:17:31 Build Failed. 4 errors, 25 warnings. (took 3s.264ms)
[error.txt](https://github.com/grblHAL/LPC176x/files/9850442/error.txt)

terjeio, 2022-10-24:
Oops, another missing update...

Change 
https://github.com/grblHAL/LPC176x/blob/24fccf1c974a4e19114afd9f5f95400f89524eef/src/tmc_uart.c#L317
to

`static void add_uart_pin (xbar_t *gpio, void *data)`
and

https://github.com/grblHAL/LPC176x/blob/24fccf1c974a4e19114afd9f5f95400f89524eef/src/tmc_uart.c#L375
to

`hal.enumerate_pins(true, add_uart_pin, NULL);`

as a quick fix.

sn0wrider, 2022-10-24:
Super!! Compiled )))))

closed 2022-10-24 by sn0wrider