grblHAL/LPC176x #31: Unable to compile SKR 1.4 Turbo with TM2209

Fires04, 2022-10-26:
Hello I try to compile the firmware in MCUXpresso IDE.  But I getting error in build. If I comment the #define TRINAMIC_ENABLE 2209 the build is ok. 

Error:
`
../src/tmc_uart.c: In function 'if_init':
../src/tmc_uart.c:375:34: warning: passing argument 2 of 'hal.enumerate_pins' from incompatible pointer type [-Wincompatible-pointer-types]
  375 |         hal.enumerate_pins(true, add_uart_pin);
      |                                  ^~~~~~~~~~~~
      |                                  |
      |                                  void (*)(xbar_t *)
../src/tmc_uart.c:375:34: note: expected 'pin_info_ptr' {aka 'void (*)(xbar_t *, void *)'} but argument is of type 'void (*)(xbar_t *)'
../src/tmc_uart.c:375:9: error: too few arguments to function 'hal.enumerate_pins'
  375 |         hal.enumerate_pins(true, add_uart_pin);
      |         ^~~
At top level:
../src/tmc_uart.c:83:24: warning: 'outputpin' defined but not used [-Wunused-variable]
   83 | static output_signal_t outputpin[] = {
      |                        ^~~~~~~~~
make[1]: *** [src/subdir.mk:66: src/tmc_uart.o] Error 1
make[1]: *** Waiting for unfinished jobs....
Finished building: ../src/usb_serial.c
 
Finished building: ../src/serial.c
 
Finished building: ../src/driver.c
 
make: *** [makefile:46: all] Error 2
"make -r -j8 all" terminated with exit code 2. Build might be incomplete.
`

my_machine.h
`// NOTE: Only one board may be enabled!
// If none is enabled pin mappings from generic_map.h will be used
//#define SMOOTHIEBOARD
//#define BOARD_RAMPS_16
//#define BOARD_BTT_SKR_13
#define BOARD_BTT_SKR_14_TURBO
//#define BOARD_BTT_SKR_E3_TURBO  // With onboard Trinamic TNMC2209 drivers. NOTE: not verified!
//#define BOARD_MKS_SBASE_13
//#define BOARD_MY_MACHINE // Add my_machine_map.h before enabling this!

// Configuration
// Uncomment to enable, for some a value > 1 may be assigned, if so the default value is shown.

#ifndef USB_SERIAL_CDC
//#define USB_SERIAL_CDC     1 // Comment out to use UART communication.
#endif
//#define SAFETY_DOOR_ENABLE 1 // Enable safety door input.
//#define SDCARD_ENABLE      1 // Run gcode programs from SD card, requires sdcard plugin.
//#define MPG_ENABLE         1 // Enable MPG interface. Requires serial port and one handshake pin.
//#define TRINAMIC_ENABLE 2130 // Uncomment to enable Trinamic TMC2130 driver support. NOTE: Experimental for now, currently for SKR 1.x boards only
#define TRINAMIC_ENABLE 2209 // Uncomment to enable Trinamic TMC5160 driver support. NOTE: Experimental for now, currently for SKR 1.x boards only
//#define TRINAMIC_ENABLE 5160 // Uncomment to enable Trinamic TMC5160 driver support. NOTE: Experimental for now, currently for SKR E3 Turbo board only
//#define LIMIT_MAX_ENABLE   1 // Uncomment to enable max limit input pins (when available)
//#define EEPROM_ENABLE      1 // I2C EEPROM support. Set to 1 for 24LC16(2K), 2 (64 byte pages) or 3 (32 byte pages) for larger sizes. Requires eeprom plugin.
//#define EEPROM_IS_FRAM     1 // Uncomment when EEPROM is enabled and chip is FRAM, this to remove write delay.

// If the selected board map supports more than three motors ganging and/or auto-squaring
// of axes can be enabled here.
//#define X_GANGED            1
//#define X_AUTO_SQUARE       1
#define Y_GANGED            1
#define Y_AUTO_SQUARE       1
//#define Z_GANGED            1
//#define Z_AUTO_SQUARE       1
// For ganged axes the limit switch input (if available) can be configured to act as a max travel limit switch.
// NOTE: If board map already has max limit inputs defined this configuration will be ignored.
//#define X_GANGED_LIM_MAX    1
//#define Y_GANGED_LIM_MAX    1
//#define Z_GANGED_LIM_MAX    1
`

Fires04, 2022-10-26:
Ok, if I edit the tmc_uart.c  line 375 to `hal.enumerate_pins(true, add_uart_pin,NULL);` the build is work. ?

terjeio, 2022-10-26:
See [this issue](https://github.com/grblHAL/LPC176x/issues/30) for quick fix.

Fires04, 2022-10-28:
Thanks it works like charm