grblHAL/LPC176x #37: SKR V 1.3 - Set Digital Pin 2.0 Low on reset

dpg7075, 2022-12-19:
Ive successfully set up a laser to run off of the servo pin 2.0 and compiled the firmware to work in that way, instead of the bed. My laser needed 5 volts instead of the 12 volts. However, the  digital pins on the LPC1768 are by default, as Ive read in the datasheet, set to high. This results in a voltage of +- 2.36 volts on pin 2.0 on reset. I can turn the lasermode off and on using $32, or inside lightburn, which sets the pin 2.0 voltage to 0.00, to low. 

Id like this pin to use the pulldown resistor internally to be set to low on reset. Can you point me in the direction so that i can accomplish this? Upon reset, with the 2.36volts on the 2.0 pin, it causes the laser to turn on. With a low voltage of 0, i could then just turn in on in low power mode with the button as intended. This way the laser wont just turn on when i plug in the board and i wont have to always have lightburn turn on and then turn on the lasermode to set the pin 2.0 voltage to low as is needed.

terjeio, 2022-12-20:
It can be done in the [board specific code](https://github.com/grblHAL/LPC176x/blob/master/src/btt_skr_1.x.c), or if it is a customization in a plugin. A [plugin](https://github.com/grblHAL/Templates/tree/master/my_plugin) can hook itself into `hal.driver_setup` (is called once on startup) and/or `hal.driver_reset`  (called on soft resets).

dpg7075, 2022-12-21:
Thanks for the info. Im still working on this. I cant seem to find enough info online that I can understand how to code this. So far, most examples ive studied use LPC17xx.h to blink LED's, (change pin high to low). Seems simple enough but I cant get to to work. I tried to inlcude the header from the grbl lpc project, but as Im sure you know, that didnt work. Im sure there are conflicts contained in it with your grbl hal code. I'll keep trying to figure it out. Thank you for all your work.

terjeio, 2022-12-22:
IIRC the Grbl LPC project is built using a different framework so the headers from that cannot be used.

BTW when you write pin 2.0 above is that P2.0 using the native pin designation or is it the Arduino digital pin 2.0? Note that the driver uses native designations.

To change pullup/pulldown state for a pin use `Chip_IOCON_PinMuxSet()` - see driver.c for how it is done there.

dpg7075, 2022-12-22:
Thank you for additional help. I am using P2.0. I will continue to work on it.

closed 2023-08-14 by terjeio