grblHAL/LPC176x #39: Probing not reporting

paukstelis, 2023-01-08:
SKR 1.4 Turbo. Probing operations work, but the "PRB" acknowledgement with position is not returned. I have verified that DEFAULT_NO_REPORT_PROBE_COORDINATES is commented out. The PRB line can be retrieved with `$#`.

terjeio, 2023-01-08:
What is your $10 setting? A bit in that setting controls the report availability. Use `$$=10` for details.
Also note that the default config values are now used to set initial setting values, use $RST=* to propagate them after reflashing - or just change the relevant settings.

paukstelis, 2023-01-08:
It was indeed $10. That got me all squared away. Next I'll have questions on getting sensorless homing working with the tmc5160s....

closed 2023-01-08 by paukstelis