grblHAL/LPC176x #40: Estop support

yoft, 2023-08-24:
Hi,
Is it difficult to add estop support to this driver? And what features does this add?

Thanks!

terjeio, 2023-08-24:
> Is it difficult to add estop support to this driver?

No, I'll add it in the next commit.

> And what features does this add?

It is a little less harsh than a soft reset, it will not immediately reinitialize the controller. Unlike a soft reset via the reset pin the e-stop state cannot be cleared before the input signal is deasserted (e-stop switch unlocked) so a tad safer and this is perhaps the major difference.
Note that for safety reasons one should never rely on the controller to shut down motors on a critical event, relays should be added, wired to the e-stop/reset switch, to cut power to them. See links from [this search](https://duckduckgo.com/?t=ffab&q=estop+functionality&ia=web) for more.

yoft, 2023-08-25:
Hi @terjeio,
You're a legend!

I'll wire the signal from the emergency contactor auxiliary contact.

Is it possible to also add support for HC-05 bluetooth?

Thanks again!

yoft, 2023-08-25:
Sorry, didn't mean to close, new to github issues.

terjeio, 2023-08-25:
> Is it possible to also add support for HC-05 bluetooth?

Which board? [An interrupt capable aux input is required](https://github.com/grblHAL/Plugins_Bluetooth/).

yoft, 2023-08-25:
> Which board? [An interrupt capable aux input is required](https://github.com/grblHAL/Plugins_Bluetooth/).

BTT SKR 1.3, I haven't checked which pins are interrupt capable, is it easy to find?

terjeio, 2023-08-25:
I have mapped P0.2 in the _Servo_ header and added the Bluetooth plugin in the [latest commit](https://github.com/grblHAL/LPC176x/commit/7485dfd069fec9e328696cce932c83c32879f703). Note that USB comms has to be enabled as default since the driver only supports one serial port. I have only tested with a Re-ARM board, I leave it to you to verify with the SKR board as I do not have one.

yoft, 2023-09-01:
Just an update, I've been unsuccessful in setting up the bluetooth as my usb port is faulty, leaving only the single serial port.

I'm yet to verify the estop, that'll probably be tomorrow.

I've decided to replace the board with BTT Octopus Pro STMF446, so I'll see you in that repo soon.

Thanks,

GundamModeler, 2023-10-17:
Which pin do I wire my E-Stop to?  The pin labeled 2.0 and GND in servo header?
[image]

yoft, 2023-10-17:
> Which pin do I wire my E-Stop to?  The pin labeled 2.0 and GND in servo header?
> [image]
> 

Hi,
I never got a chance to verify functionality, but I believe the estop replaces the reset pin, P0.18 in the board map. You will have to compile with

GundamModeler, 2023-10-17:
ahh thx for the fast reply, I will test in the morning!