grblHAL/LPC176x #41: Device Descriptor Request Failed

yoft, 2023-08-25:
Hi,
I'm having an issue with windows and BTT SKR 1.3.

Everything was working fine previously, the machine had been turned off for maybe a week.

Now when I plug the usb cable into the computer I get a popup saying the usb device malfunctioned.
In Device Manager it is labelled as "Unknown USB Device (Device Descriptor Request Failed)".

Is this a software issue?

I've tried 3 different cables, and the board is also currently sitting on the workbench completely disconnected, and now pwered from usb, I've also tried powering from the 24v PSU.

Thanks,

terjeio, 2023-08-25:
Hard to tell - could be a dead MCU (unlikely since the board is detected?), windows driver issue, ...
This is the driver used by my Win10 machine:

[image]

Are you able to reflash the controller?

yoft, 2023-08-25:
Hi,
SD bootloader still works, I tried renaming firmware.cur to firmware.bin, I assume this worked as it'd change back to .cur after powering it up.
Is there any other information I can provide that will help?
I appreciate your hard work!

terjeio, 2023-08-25:
SD bootloader still works, I tried renaming firmware.cur to firmware.bin, I assume this worked as it'd change back to .cur after powering it up.

Ok, this was with the same firmware used when it was working?

Since it worked a week ago and no changes was made to the controller it is either a new hardware problem with the controller or an issue with the Windows USB driver/stack. Is your  USB driver the same as I have installed?

yoft, 2023-08-25:
> Ok, this was with the same firmware used when it was working?

Yes, same firmware.

> Since it worked a week ago and no changes was made to the controller it is either a new hardware problem with the controller or an issue with the Windows USB driver/stack. Is your USB driver the same as I have installed?

I'm starting to lean towards hardware issue.
None if the driver information is available, on Windows 8 it lists everything on that page as either "unknown" or "not available". It's the same for all usb ports across two computers, I tried a brand new usb cable too, just to rule that out.
I tried compiling a build with USB_SERIAL_CDC commented out, just to see if there was anything different. No change. Recompiled with it enabled again, still no change.

yoft, 2023-08-25:
I've just tried using Smoothieware firmware, issue persists.
I'm writing this board off. 
Thank you for your help!

closed 2023-08-25 by yoft

yoft, 2023-08-25:
> I tried compiling a build with USB_SERIAL_CDC commented out

When I set up the usb-serial adapter properly, I can connect with IO Sender!
Looks like this will be the solution for now...