grblHAL/LPC176x #43: Can't seem to get SKR 1.3 + TMC 2209 to work, can anyone help?

GundamModeler, 2023-09-24:
Im trying to get this board to work with my 3018 style CNC but I can't get the motor to turn. It will stutter 3 times when i try to move the X axis and then just stop until unplug the power and replug it back in. I had a pretty hard time learning how to compile but manage to get a firmware.bin and uplaoded to the board but Im unsure if I did it correctly.
Here is what I changed 
[image]

Here is what I get in Universal GCode Sender

I had to change the following to get no alarms
$5=7
$14=3
$17=4

everything else I think is default.

[VER:1.1f.20230919:]
[OPT:VNMSL,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:LCP1769]
[DRIVER VERSION:230919]
[BOARD:https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3]
[PLUGIN:Trinamic v0.12]
ok
>>> $$
$0 = 10.0    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 7    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$9 = 1   
$10 = 511    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$14 = 3   
$15 = 0   
$16 = 0   
$17 = 4   
$18 = 0   
$19 = 0   
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.0    (Homing locate feed rate, mm/min)
$25 = 500.0    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$28 = 0.100   
$29 = 0.0   
$30 = 1000.000    (Maximum spindle speed, RPM)
$31 = 0.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$33 = 5000.0   
$34 = 0.0   
$35 = 0.0   
$36 = 100.0   
$37 = 0   
$39 = 1   
$40 = 0   
$43 = 1   
$44 = 4   
$45 = 3   
$46 = 0   
$62 = 0   
$63 = 3   
$64 = 0   
$65 = 0   
$100 = 250.000    (X-axis travel resolution, step/mm)
$101 = 250.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
$140 = 0   
$141 = 0   
$142 = 0   
$150 = 16   
$151 = 16   
$152 = 16   
$180 = 25.0   
$181 = 25.0   
$182 = 25.0   
$190 = 500.0   
$191 = 500.0   
$192 = 500.0   
$200 = 22.0   
$201 = 22.0   
$202 = 22.0   
$210 = 50   
$211 = 50   
$212 = 50   
$220 = 22.0   
$221 = 22.0   
$222 = 22.0   
$338 = 0   
$339 = 0   
$341 = 0   
$342 = 30.0   
$343 = 25.0   
$344 = 200.0   
$345 = 200.0   
$346 = 1   
$384 = 0   
$398 = 35   
$481 = 0   
$486 = 0

terjeio, 2023-09-24:
Try with setting the motor current to reasonable values: $140-$142, settings are mA RMS.
You may have to tune other [Trinamic driver settings](https://github.com/grblHAL/Plugins_motor/) as well, e.g. the microstepping values.

FYI if you are using a Windows computer you may try [ioSender](https://github.com/terjeio/ioSender) for configuration and motor tuning. It has an advanced UI for both.

terjeio, 2023-09-24:
Oops, I forgot $338 - set it to 7 to enable UART communication with the drivers.

GundamModeler, 2023-09-24:
Wow I didn't expect a reply for a while or at all tbh, Thank you! 

So I set my M140=800  and $338=7 and it works  for my smaller Nema 17 42mm motor, It spins quietly as expected but on my bigger unknown model# Nema 17 60mm ( It came with my chinese 4018) it just kinda jiggles/stutters back and forth rapidly. 

I tried up to 1200mah and it still does the same thing only more intensely, and Im afraid to push it any higher as the heatsink gets kinda hot. I see everyone using around 1000mah. All I know about the motor is that it's a Nema 17 42x42x60mm with 1.7ohm on both coils

At this point im pretty sure I need to tune the tmc2209, Im thinking theres is probably something that is causing it to stall. I playing around with ioSender and it looks really cool, do you have a guide on how to tune the tmc2209s? Also is there a version that runs on Raspberry Pi?

terjeio, 2023-09-24:
You need to set the current for Y ($141) and Z ($142) - both are 0 in the M122 report above. Or did you try the bigger motors on X as well?
I see that some of the Trinamic settings values are wrong/not set to default values, try resetting them with `$RST=&` and configure from default values.

> do you have a guide on how to tune the tmc2209s?

No, currently I have [linked to the Trinamic datasheets](https://github.com/grblHAL/Plugins_motor/#339---sensorless-homing-enable) for that.

> [ioSender] Also is there a version that runs on Raspberry Pi?

No, it is Windows only.

GundamModeler, 2023-09-24:
Ok after $RST=& ands renabling UART I get these values. I currently have 3x of the smaller 48mm nema 17 motors hooked up but only the X axis works while the other  don't do anything.
[image]

terjeio, 2023-09-24:
What is your $4 setting? Typically this should be 7 (all enable pins inverted).

The Z motor is log odd "olb" means the motor B windings are "open" (not connected). "otpw" is overtemperature pre warning.
Are both Y and Z drivers hot? Z very hot? Be careful not to burn your fingers!

What happens when you swap X and Y drivers? Same issue?

GundamModeler, 2023-09-24:
Wow changing $4=7 seem to have fix the problems and now all motors spin when jogged. Regarding the overtemperature, yes it was hot and I couldn't figure out why until I inspected the connectors up close and found the wires was frayed and was shorting. I replaced the connector and now it works without any problems.  

This also lead to me inspecting the motor on the larger 60mm motor and discovered the wiring was different. I accidentally used the ender 3 cables and didn't realize the middle wires were twisted. I switch the wires and now the motors works.

Thank you for all the help, I was so frustrated and ready to buy a new board but now it works great. 
[image]

closed 2023-10-02 by terjeio