grblHAL/LPC176x #50: Probe settings not propogating?

paukstelis, 2024-08-22:
Board: BTT SKR Turbo 1.4

With both DEFAULT_PROBE_SIGNAL_INVERT and DEFAULT_PROBE_SIGNAL_DISABLE_PULLUP set to `On` I expected to have $6 and $19 available. However, these don't appear to be available (older grblHal version I have running  on a different board does have these and I can use the probe just fine). I haven't been able to get the probe to trigger at all, even simply by jumpering the defined probe pin.

paukstelis, 2024-08-23:
It seems like this is related to trying to change the probe port and pin. If I try to use Port 1 and any number of pins it will not work. But the default port and pin (0.2) does work. it would be more convenient for this particular to probe to use any of the endstop pins since they already have 5V and GND in the header as well. Using the Probe port means having to split between two connectors.

terjeio, 2024-08-24:
> If I try to use Port 1 and any number of pins it will not work. 

Port 1 can be used when I test with a SKR 1.3 board.
What is the voltage on the probe pin when the probe is not connected? If 0V you may have disabled the pullup by setting `$19=1`?

paukstelis, 2025-04-04:
I'm swinging back around to this because I can now not get the probe to be compiled at all. No matter what I change for the port and pin I get this in the header:

`Recv: [VER:1.1f.20250329:]
Recv: [OPT:NMDSL,100,1024,5,0]
Recv: [AXS:5:XYZAB]
Recv: [NEWOPT:ENUMS,RT+,NOPROBE,EXPR,SED]
Recv: [FIRMWARE:grblHAL]
Recv: [SIGNALS:RHS]
Recv: [NVS STORAGE:*FLASH]
Recv: [DRIVER:LCP1769]
Recv: [DRIVER VERSION:250327]
Recv: [BOARD:BTT SKR V1.4 Turbo]
Recv: [AUX IO:1,3,0,0]
Recv: [PLUGIN:Trinamic v0.27]`

paukstelis, 2025-04-04:
There was clearly some conflict coming from btt_skr_1.4_turbo_map.h file that I had been using for previous versions. Using a clean map file seems to get it  working again.