grblHAL/LPC176x #52: mks sbase v1.3 flashing help

dasbrow, 2025-05-10:
I've managed to get this to compile and produce a axf file, but not sure how to flash a sbase board with the file. I tried to covert it to a bin file via mcuxpresso and it's didn't work.

Thanks in advance.

terjeio, 2025-05-10:
Seems I have only enabled conversion to bin for the bootloader configuration (BL0x4000), right click on the project root and remove comments (the #) from the objcopy and checksum steps:

[image]

dasbrow, 2025-05-12:
Thanks for the reply. I did a boot loaded update (copied it to SD card and reset board), now the boards no long work. Likely just a bad flash. I've ordered an ICSP debugger from NXP to hopefully flash the board correctly. Should have it tomorrow. If I am going about this the wrong way let me know.

terjeio, 2025-05-12:
See the [ReadMe](https://github.com/grblHAL/LPC176x/blob/master/README.md) entry for 2020-11-21 for how to create a bootloader version. The _Debug_ and _Release_ configurations will _not_ work when flashed via the SD card.

dasbrow, 2025-05-12:
Thanks, got it working.

closed 2025-05-12 by dasbrow